espressif/esp-drone #2: Is it possible to use this hardware/software for a car ?

happytm, 2020-07-15:
I wanted to use your hardware and software for toy car using 4 - 24 VDC motors. Is it possible?

Thanks.

leeebo, 2020-07-15:
Yes , you can use the firmware and APP to your toy car project, but some coding work need to do :

1. remove the sensors and related task ( maybe you don't neea any sensors ).
2. remove stabilize task and kalman task ... ( cars don't need 
 attitude calculate and angle control).
3. send rypt command directly to motors through power distribution. 

Good luck to you ！

happytm, 2020-07-15:
Thank you for you quick response. 

You are right I do not need any sensors for stabilization except to control car steering left and right to  make it turn and reverse the car staight.

leeebo, 2020-07-16:
My pleasure to help you, you can add new issue in detail during your development process.

leeebo, 2020-07-19:
You can close this issue if the answer makes sense.

leeebo, 2020-07-19:
You can close this issue if the answers make sense.

commit 686ecdd6 (2020-08-07): Merge branch 'bugfix/log_cause_unexpected_shutdown' into 'master'

bugfix(log): fix log plotter bug

Closes #2

See merge request ae_group/esp-college/esp-drone!17
commit a8c2ffe9 (2020-08-07): bugfix(log): fixed bug when use log plotter

upgrade assertFail print level to show erro line

delete cfclient config cache in /home/username/.config/cfclient can fix the problem.

Close #2
closed 2020-07-19 by happytm