espressif/esp-drone #5: esp-drone  compile error

AzollaGit, 2020-08-25:
[image]
大佬好：
我采用最新 esp-idf - mater分支， idf 例程编译没有问题，esp-drone 近百个这样类似的错误，不知道为什么，希望解答！

谢谢！

leeebo, 2020-08-25:
Clean your project then execute the following commands step by step

```
cd esp-drone
rm -r build sdkconfig sdkconfig.old
idf.py set-target esp32s2
idf.py build
```

AzollaGit, 2020-08-25:
> Clean your project then execute the following commands step by step
> 
> ```
> cd esp-drone
> rm -r build sdkconfig sdkconfig.old
> idf.py set-target esp32s2
> idf.py build
> ```

你好！ 我按照你给的指令依次执行，还是报相同的错误。 esp-drone 是用git clone 下来的。

leeebo, 2020-08-25:
```
git status

leeebo, 2020-08-25:
Please add more details to reproduce the failure:

```
OS:{windows 7 8 10/ubuntu/Mac ...?}
esp-idf version: {branch} {commitID} {any change?}{submodule ready?} 
toolchain：{all update？} //you can rerun install.xx to update it
```

AzollaGit, 2020-08-26:
> 请添加更多详细信息以重现故障：
> 
> ```
> OS:{windows 7 8 10/ubuntu/Mac ...?}
> esp-idf version: {branch} {commitID} {any change?}{submodule ready?} 
> toolchain：{all update？} //you can rerun install.xx to update it
> ```

你好！
>OS: ubuntu 18.04 TLS
>esp-idf:  version: { master }  {Merge: 9e45dca65 ee135b8e2} 
 {commit 6c17e3a64c02eff3a4f726ce4b7248ce11810833 (HEAD -> master, origin/master, origin/HEAD) }
有执行 git pull           git submodule update --init --recursive 
>toolchain: 参考官方教程执行  cd esp-idf        ./install.sh  
测试 $IDF_PATH/examples/get-started/hello_world    idf.py set-target esp32s2  编译通过。
谢谢！

leeebo, 2020-08-26:
Hi your code seems in a wrong place, have you changed the directory structure ?

[image]

AzollaGit, 2020-08-26:
> 嗨，你的代码似乎在一个错误的地方，你改变了目录结构？
> 
> [image]

Hi 
I didn't make any changes to Git clone.   An error will still be reported if the following instructions are executed.
>idf.py set-target esp32s2
>idf.py build
[image]

AzollaGit, 2020-08-27:
你好：
>esp-idf { commit 98d5b5dfd94c5dbc0c50be8d100651e118b56a0f (HEAD -> origin_v4.2, origin/release/v4.2, espressif/release/v4.2) }
当使用： idf.py set-target esp32  编译esp-drone 可以通过。
使用 idf.py set-target esp32s2 会报一下错误；请问是esp32s2不支持吗？ 开源硬件使用的是esp32s2，相信应该是支持的吧。
[image]

leeebo, 2020-08-27:
> 你好：
> 
> > esp-idf { commit 98d5b5dfd94c5dbc0c50be8d100651e118b56a0f (HEAD -> origin_v4.2, origin/release/v4.2, espressif/release/v4.2) }
> > 当使用： idf.py set-target esp32  编译esp-drone 可以通过。
> > 使用 idf.py set-target esp32s2 会报一下错误；请问是esp32s2不支持吗？ 开源硬件使用的是esp32s2，相信应该是支持的吧。
> > [image]

Yes, esp32s2 is also supported. The `pxCurrentTCB` bug comes from esp-idf , which has been fixed in master branch. You can refer [commit: 600665a](https://github.com/espressif/esp-idf/commit/600665a5638b36c0b6defed56614172ce67ac4ae#diff-b0927f18912256b20af5dd812a81ca11) to modify your esp-idf code.

AzollaGit, 2020-08-30:
> > 你好:
> > > esp-idf = 提交 98d5b5dfd94c5dbc0c50be8d100651e118b56a0f （HEAD -> origin_v4.2， 原点/发布/v4.2， espressif/发布/v4.2）  = idf.py 设定目标 esp32esp- drone = [idf.py设定目标 esp32s2];][esp32s2]？
> > > [image]
> 
> 是，esp32s2 也受支持。错误来自esp-idf，它已被固定在主分支中。您可以引用[提交：600665a](https://github.com/espressif/esp-idf/commit/600665a5638b36c0b6defed56614172ce67ac4ae#diff-b0927f18912256b20af5dd812a81ca11)来修改您的 esp-idf 代码。`pxCurrentTCB`

谢谢解答，问题已经解决了，顺利编译通过。 非常感谢！

leeebo, 2020-08-31:
Hi @AzollaGit, would you please add your cmake and ninja version below to help us find the bug.

you can run:

```
cmake --version
ninja --version
```

to get the installed version in your PC.

leeebo, 2020-08-31:
@AzollaGit We have added a [new commit](https://github.com/espressif/esp-drone/commit/b715cc7e75763427803d78f582747c0ea636b64b) to fix the bug, you can checkout to master to try.

AzollaGit, 2020-09-01:
> @AzollaGit我们添加了一[个新的提交](https://github.com/espressif/esp-drone/commit/b715cc7e75763427803d78f582747c0ea636b64b)来修复错误， 您可以签出您的 ESP - idf 以掌握以尝试它。

你好！
cmake version 3.17.2
ninja version 1.8.2
得知你有新的提交；我在esp-drone 目录执行 git pull  后，会以下报错，我查询是esp-idf/components/esp_adc_cal 下
[image]

[image]

leeebo, 2020-09-01:
> > @AzollaGit我们添加了一[个新的提交](https://github.com/espressif/esp-drone/commit/b715cc7e75763427803d78f582747c0ea636b64b)来修复错误， 您可以签出您的 ESP - idf 以掌握以尝试它。
> 
> 你好！
> cmake version 3.17.2
> ninja version 1.8.2
> 得知你有新的提交；我在esp-drone 目录执行 git pull 后，会以下报错，我查询是esp-idf/components/esp_adc_cal 下
> [image]
> 
> [image]

You need to update your esp-idf to latest commit.

leeebo, 2020-09-07:
Could you please close the issue if your problem got resolved.

closed 2022-11-24 by zhanzhaocheng