espressif/esp-drone #7: esp-drone compile error

QinYUN575, 2020-08-26:
# esp-drone compile error

## Development environment
- OS: Ubuntu 20.04 focal(on the Windows Subsystem for Linux)
- esp-idf version: 
  master {commit c77c4ccf6c43ab09fd89e7c907bf5cf2a3499e3b (HEAD -> master, origin/master, origin/HEAD)} 
 {No changes}
{All sub-modules are successfully pulled} 
toolchain：Install using install.sh in the SDK directory(esp-2020r2-8.2.0)
------

## Error msg

'undefined reference to `sleepus''

[image]
### Possible reasons: IDF version is inconsistent with the project dependent version (**the project does not specify the dependent IDF version**)
-----
Looking forward to your reply, thanks🙇‍

QinYUN575, 2020-08-26:
IDF example can be compiled normally

leeebo, 2020-08-26:
We checkout to the commit `c77c4ccf6c` as you described above, both `esp32` and `esp32s2` can build successfully. (using ubuntu18.04 and windows 10).

Your build system seems not correctly parse [the CMakeLists.txt in pmw3901 dir](https://github.com/espressif/esp-drone/blob/master/components/drivers/spi_devices/pmw3901/CMakeLists.txt#L3):

1.  you can modify `PRIV_REQUIRES` to `REQUIRES` then try again
2.  if it doesn't work, you can checkout your IDF to branch `release/v4.2` then have a try.

leeebo, 2020-08-26:
Q: IDF version is inconsistent with the project dependent version (the project does not specify the dependent IDF version)

A: Yes we haven't specify the IDF version and commitID, because this project is on developing. On the other hand, IDF support for esp32s2 is not stable now. You can use the latest IDF master version or release/v4.2, when you find problems you can add issues to related repo, we will try to fix them.

QinYUN575, 2020-08-30:
According to the method provided by #5   switch to the release/v4.2 branch, you can successfully compile for the ESP32 project.
Thank! 🙇‍

closed 2020-08-30 by QinYUN575

leeebo, 2020-08-31:
@QinYUN575 Hi, would you please add your cmake and ninja version below to help us find the bug.

you can run:

```
cmake --version
ninja --version
```

to get the installed version in your PC.

leeebo, 2020-08-31:
Hi @QinYUN575 , we have added a [new commit](https://github.com/espressif/esp-drone/commit/b715cc7e75763427803d78f582747c0ea636b64b) to fix the bug, you can checkout ESP-IDF to master to try

QinYUN575, 2020-08-31:
@leeebo Thank! 
Now, switch the esp-drone version to [commit:b715cc7](https://github.com/espressif/esp-drone/commit/b715cc7e75763427803d78f582747c0ea636b64b)/or master ,compile esp32 esp32s2 and all pass.
---
cmake --version
>> cmake version 3.17.2
ninja --version
>>1.10.0