espressif/esp-drone #9: compile fail

yuhuilian, 2020-10-28:
I try to follow the steps to compile  the latest code of esp-done:
`idf.py set-target esp32s2`
`idf.py build`

But I got the flowing error, and compile failed:
`
In file included from ../components/drivers/i2c_bus/include/i2c_drv.h:4,
                 from ../components/drivers/i2c_bus/include/i2cdev.h:30,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform_user_data.h:40,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1x.h:34,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform.h:33,
                 from ../components/drivers/i2c_devices/vl53l1/core/inc/vl53l1_core.h:72,
                 from ../components/drivers/i2c_devices/vl53l1/core/src/vl53l1_api_debug.c:71:
H:/Workspace/YuHuiLian/ESP32/esp-idf/components/freertos/include/freertos/FreeRTOS.h:963:27: error: conflicting types for 'SemaphoreHandle_t'
  #define xSemaphoreHandle SemaphoreHandle_t
                           ^~~~~~~~~~~~~~~~~
../components/platform/stm32_legacy.h:25:15: note: in expansion of macro 'xSemaphoreHandle'
 typedef void *xSemaphoreHandle;
               ^~~~~~~~~~~~~~~~
In file included from ../components/drivers/i2c_bus/include/i2c_drv.h:5,
                 from ../components/drivers/i2c_bus/include/i2cdev.h:30,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform_user_data.h:40,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1x.h:34,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform.h:33,
                 from ../components/drivers/i2c_devices/vl53l1/core/inc/vl53l1_core.h:72,
                 from ../components/drivers/i2c_devices/vl53l1/core/src/vl53l1_api_debug.c:71:
H:/Workspace/YuHuiLian/ESP32/esp-idf/components/freertos/include/freertos/semphr.h:37:23: note: previous declaration of 'SemaphoreHandle_t' was here
 typedef QueueHandle_t SemaphoreHandle_t;
                       ^~~~~~~~~~~~~~~~~
In file included from ../components/drivers/i2c_bus/include/i2c_drv.h:4,
                 from ../components/drivers/i2c_bus/include/i2cdev.h:30,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform_user_data.h:40,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1x.h:34,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform.h:33,
                 from ../components/drivers/i2c_devices/vl53l1/core/inc/vl53l1_core.h:72,
                 from ../components/drivers/i2c_devices/vl53l1/core/src/vl53l1_api_debug.c:71:
H:/Workspace/YuHuiLian/ESP32/esp-idf/components/freertos/include/freertos/FreeRTOS.h:962:23: error: conflicting types for 'QueueHandle_t'
  #define xQueueHandle QueueHandle_t
                       ^~~~~~~~~~~~~
../components/platform/stm32_legacy.h:27:15: note: in expansion of macro 'xQueueHandle'
 typedef void *xQueueHandle;
               ^~~~~~~~~~~~
In file included from H:/Workspace/YuHuiLian/ESP32/esp-idf/components/freertos/include/freertos/semphr.h:35,
                 from ../components/drivers/i2c_bus/include/i2c_drv.h:5,
                 from ../components/drivers/i2c_bus/include/i2cdev.h:30,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform_user_data.h:40,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1x.h:34,
                 from ../components/drivers/i2c_devices/vl53l1/include/vl53l1_platform.h:33,
                 from ../components/drivers/i2c_devices/vl53l1/core/inc/vl53l1_core.h:72,
                 from ../components/drivers/i2c_devices/vl53l1/core/src/vl53l1_api_debug.c:71:
H:/Workspace/YuHuiLian/ESP32/esp-idf/components/freertos/include/freertos/queue.h:48:34: note: previous declaration of 'QueueHandle_t' was here
 typedef struct QueueDefinition * QueueHandle_t;
                                  ^~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
`

leeebo, 2020-11-03:
Please add your ESP_IDF commit message, we suggest you update to the latest version in master branch.

YuAIoT, 2020-11-03:
@leeebo 
The ESP_IDF commit message is:

commit 64cb09893eac6aad56a40bacfc17d8fd72b854e9 (origin/master, origin/HEAD)
Merge: c54af56 cc5486b
Author: Li Bo <libo@espressif.com>
Date:   Mon Sep 21 11:04:34 2020 +0800

    Merge branch 'bugfix/component_name_conflict' into 'master'

    bugfix(hal): component name conflict with esp-idf

    See merge request ae_group/esp-college/esp-drone!28

**After commenting out some code in /stm32_legacy.h like below,  compiling sucseed.**

diff --git a/components/platform/stm32_legacy.h b/components/platform/stm32_legacy.h
index 56316fc..697cfab 100644
--- a/components/platform/stm32_legacy.h
+++ b/components/platform/stm32_legacy.h
@@ -22,10 +22,10 @@ typedef struct {
 //freertosconfig
 typedef void DMA_Stream_TypeDef;
 typedef int DMA_InitTypeDef;
-typedef void *xSemaphoreHandle;
-typedef void *SemaphoreHandle_t;
-typedef void *xQueueHandle;
-typedef void *QueueHandle_t;
+//typedef void *xSemaphoreHandle;
+//typedef void *SemaphoreHandle_t;
+//typedef void *xQueueHandle;
+//typedef void *QueueHandle_t;
 typedef void *TaskHandle_t;
 typedef void TIM_TypeDef;
 typedef void TIM_OCInitTypeDef;

leeebo, 2020-11-03:
Thanks, we will fix it then clean the code.

leeebo, 2020-11-03:
@YuAIoT Bug fixed, please close this issue.

closed 2023-07-28 by leeebo