espressif/esp-drone #11: Linking ERROR

maxic62, 2021-01-20:
Hello,
I'm facing this error
[linkErrorLog.txt](https://github.com/espressif/esp-drone/files/5843639/linkErrorLog.txt)
[image]

platform: windows 10
target: esp32

Could you help please ?
Thanks

leeebo, 2021-01-21:
It looks like you made a mistake in this step:

Insert below code into the end of `.flash.rodata` section, which detailed in the [Modify ESP32/ESP32-S2 Link Script](https://docs.espressif.com/projects/espressif-esp-drone/en/latest/getespidf.html#modify-esp32-esp32-s2-link-script)

```
/* Parameters and log system data */
 _param_start = .;
 KEEP(*(.param))
 KEEP(*(.param.*))
 _param_stop = .;
 . = ALIGN(4);
 _log_start = .;
 KEEP(*(.log))
 KEEP(*(.log.*))
 _log_stop = .;
 . = ALIGN(4);
```

If this doesn't solve your problem, you can update to the master branch then try again.

maxic62, 2021-01-21:
Hello,

Sorry, my mistake, I wrote this code at the complete end of the file instead of the end of the section.

Thanks for the support! It compiles and link now ;)
Have a great day

closed 2021-01-21 by maxic62