espressif/esp-drone #12: Compile error: Missing header files

physiii, 2021-02-12:
I get an error when compiling saying I'm missing _stm32_legacy.h_ however I can see the file in _components/platform_.

When commented out, I get an error on the next header _platform.h_.

Adding "./components/platform" to CMakeLists.txt didn't fix it.

Any suggestions?

> $ make all
> Toolchain path: /home/andy/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> Toolchain version: esp-2020r3
> Compiler version: 8.4.0
> Python requirements from /home/andy/esp/esp-idf/requirements.txt are satisfied.
> 
> App "ESPDrone" version: c73a879-dirty
> CC build/main/main.o
> /home/andy/devices/drone/main/main.c:30:10: fatal error: stm32_legacy.h: No such file or directory
>  #include "stm32_legacy.h"
>           ^~~~~~~~~~~~~~~~
> compilation terminated.
> make[1]: *** [/home/andy/esp/esp-idf/make/component_wrapper.mk:292: main.o] Error 1
> make: *** [/home/andy/esp/esp-idf/make/project.mk:635: component-main-build] Error 2

leeebo, 2021-02-18:
Hi @physiii , please use CMake instead. 
```
idf.py set-target esp32s2
idf.py build flash monitor
```
For more details you can refer [build-system](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html)

physiii, 2021-02-18:
Hi @leeebo  thank you for the reply.

I am using the latest esp-idf and ran idy.py fuillclean, but get the following error:

> $ idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory /home/andy/devices/esp-drone/build
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable ESPDrone.elf
FAILED: ESPDrone.elf 
: && /home/andy/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++  -mlongcalls   CMakeFiles/ESPDrone.elf.dir/project_elf_src_esp32s2.c.obj  -o ESPDrone.elf  -Wl,--start-group  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/motors/libmotors.a  esp-idf/platform/libplatform.a  esp-idf/utils/libutils.a  esp-idf/i2c_bus/libi2c_bus.a  esp-idf/deck/libdeck.a  esp-idf/mpu6050/libmpu6050.a  esp-idf/ms5611/libms5611.a  esp-idf/hmc5883l/libhmc5883l.a  esp-idf/pmw3901/libpmw3901.a  esp-idf/vl53l1/libvl53l1.a  esp-idf/vl53l0/libvl53l0.a  esp-idf/led/libled.a  esp-idf/eeprom/libeeprom.a  esp-idf/dsp_lib/libdsp_lib.a  esp-idf/wifi/libwifi.a  esp-idf/adc/libadc.a  esp-idf/crazyflie/libcrazyflie.a  esp-idf/buzzer/libbuzzer.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/andy/devices/esp-drone/build/ESPDrone.map  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/console/libconsole.a  esp-idf/json/libjson.a  esp-idf/buzzer/libbuzzer.a  esp-idf/motors/libmotors.a  esp-idf/platform/libplatform.a  esp-idf/utils/libutils.a  esp-idf/i2c_bus/libi2c_bus.a  esp-idf/deck/libdeck.a  esp-idf/mpu6050/libmpu6050.a  esp-idf/ms5611/libms5611.a  esp-idf/hmc5883l/libhmc5883l.a  esp-idf/pmw3901/libpmw3901.a  esp-idf/vl53l1/libvl53l1.a  esp-idf/vl53l0/libvl53l0.a  esp-idf/led/libled.a  esp-idf/eeprom/libeeprom.a  esp-idf/wifi/libwifi.a  esp-idf/adc/libadc.a  esp-idf/crazyflie/libcrazyflie.a  esp-idf/motors/libmotors.a  esp-idf/platform/libplatform.a  esp-idf/utils/libutils.a  esp-idf/i2c_bus/libi2c_bus.a  esp-idf/deck/libdeck.a  esp-idf/mpu6050/libmpu6050.a  esp-idf/ms5611/libms5611.a  esp-idf/hmc5883l/libhmc5883l.a  esp-idf/pmw3901/libpmw3901.a  esp-idf/vl53l1/libvl53l1.a  esp-idf/vl53l0/libvl53l0.a  esp-idf/led/libled.a  esp-idf/eeprom/libeeprom.a  esp-idf/wifi/libwifi.a  esp-idf/adc/libadc.a  esp-idf/crazyflie/libcrazyflie.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/dsp_lib/libdsp_lib.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  e [... cut ...]
/home/andy/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0x0): undefined reference to `_param_start'
/home/andy/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0xc): undefined reference to `_param_stop'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

leeebo, 2021-02-18:
Hi @physiii please can refer #11

physiii, 2021-02-18:
That worked, thank you!

closed 2021-02-18 by physiii