espressif/esp-drone #13: Param set error with cfclient

NelsenEW, 2021-02-27:
I am currently using esp-drone branch from crazyflie-lib-python, crazyflie-clients-python and January master version for the esp-drone firmware.

I was able to communicate with a custom esp32 drone for logging, visualization and plotting.
[image]

In addition, the initial parameters given on the cfclients looks correct, as I have double checked it with the firmware.
However, when I tried to change the value of the parameters with RW flag, there is red label on the entire row for that particular parameter.
[image]

Is there any way to fix this?

leeebo, 2021-02-27:
Hi @NelsenEW, thanks for your report. This is a known issue usually caused by UDP packet loss, you can try to clear the cache of cfclient and reconnect to Wi-Fi for testing. BTW, when you try to modify multiple parameters, please wait for the first complete (acked) before doing another one...

As the esp32-s2 USB worked now, I will add USB support for a more stable connection in the near future.

NelsenEW, 2021-02-27:
Okay, I am currently using an esp32s1 not s2, I will try the method suggested. Anyway, (not related to issue) do you have plan to develop camera interface code for s2?

leeebo, 2021-02-27:
@NelsenEW The current esp-drone can't handle flight control and the camera works simultaneously, because of the limited computing and memory resource of single-core esp32-s2. Future esp32-s3 maybe can handle that, I will try it then. There is no chip named esp32s1, I guess you're talking about esp32, which has no enough pins to add a DVP camera.

NelsenEW, 2021-02-27:
I have developed a custom hardware with esp32 and camera interface, it is known that esp32-s from AI-thinker can interface with camera. I have tested my custom board with camera, and it is working great using the code `CameraWebServer.ino`. Similarly, the esp-drone firmware is also working with the custom board. So I am wondering if it is possible to do that with esp32-s/ esp32.

NelsenEW, 2021-02-27:
I have tried deleting 3 different cache folders(__pycache__) located in cfclient, utils, and UI. Unfortunately, I still encounter the same error. Is there anything that I missed?

leeebo, 2021-02-27:
@NelsenEW Which development board did you use?

NelsenEW, 2021-02-27:
We are using custom board with ESP32-S processor (the one on AI-thinker), but I have tested with the esp android app and it is able to fly.

leeebo, 2021-03-01:
@NelsenEW In dir `~/.config/cfclient`, you can delete it to empty.

NelsenEW, 2021-03-01:
@leeebo I just tried deleting the cache folder inside cfclient, apparent it does not work. I will take a look into the repo commit that you mention and let you know if everything works well. Thanks!

closed 2023-07-28 by zhanzhaocheng