espressif/esp-drone #14: abort() was called at PC 0x4002450f on core 0

xmeow, 2021-03-03:
Log first:
```
[0;32mI (3484) COMM: platformserviceTest = 1 [0m
[0;32mI (3488) COMM: consoleTest = 1 [0m
[0;32mI (3492) COMM: paramTest = 1 [0m
[0;32mI (3496) SYS: commTest = 1 [0m
[0;32mI (3499) SYS: commanderTest = 1 [0m
[0;32mI (5317) SYS: stabilizerTest = 1 [0m
[0;32mI (5317) SYS: estimatorKalmanTaskTest = 1 [0m
[0;32mI (5317) SYS: memTest = 1 [0m
[0;32mI (5318) STAB: Wait for sensor calibration...
[0m
[0;32mI (5323) SYS: systemStart ! selftestPassed = 1[0m

abort() was called at PC 0x400244ef on core 0

Backtrace:0x4002b1e6:0x3ffe5c70 0x4002ba9d:0x3ffe5c90 0x4003325a:0x3ffe5cb0 0x400244ef:0x3ffe5d20 0x4002461d:0x3ffe5d50 0x400fb43e:0x3ffe5d70 0x400fe49d:0x3ffe6080 0x4010741b:0x3ffe60b0 0x40031cf2:0x3ffe60e0 0x400ed8ad:0x3ffe6130 0x400ed8dd:0x3ffe61b0 0x400270e7:0x3ffe6200 0x400271c1:0x3ffe6260 0x40028c28:0x3ffe6280 0x400d9f6b:0x3ffe62a0 0x40034c31:0x3ffe62e0
```

My hardware target is ESP-Drone II(ESP32S2), and followed all the dev guide from the doc site. The build process is smooth, but crashed right after everything is calibrated ok.

I am use latest esp-idf v4.4-dev.

Here is the map file from the build, please advice if more information should be provide. Thank you very much~

[ESPDrone.map.zip](https://github.com/espressif/esp-drone/files/6074113/ESPDrone.map.zip)

leeebo, 2021-03-03:
@xmeow Hi, this is a known bug from the latest ESP-IDF (We are working on that) .

Please checkout your ESP-IDF to a release branch like `release/v4.2` to fix.

xmeow, 2021-03-03:
Thanks, branch release/v4.2 works

closed 2021-03-03 by xmeow