espressif/esp-drone #15: Upload BOM and CPL

evilrat, 2021-06-09:
please provide BOM and CPL files for ordering pub with sod component presolderd

leeebo, 2021-06-10:
@evilrat  Hi, please check here for hardware details https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/src

evilrat, 2021-06-10:
> @evilrat Hi, please check here for hardware details https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/src

unable to open .pcb file please specify software to open it....gerber file doesn't contain any BOM or CPL...I am placing order on JLC PCB....I don't have any equipment to solder smd component....so I want to get it done by JLC PCB SMT Assembly service

leeebo, 2021-06-10:
@evilrat You can use the PADS free viewer to view databases and designs from PADS Logic or PADS Layout.

evilrat, 2021-06-10:
unable to generate CPL file for pick and place.....please help....or please upload CPL File it will be very helpful

leeebo, 2021-07-29:
@evilrat Sorry for being late, please check https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/Bom for the bom files.

evilrat, 2021-07-29:
> @evilrat Sorry for being late, please check https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/Bom for the bom files.

And pick and place file too...please....🙏

leeebo, 2021-07-29:
> > @evilrat Sorry for being late, please check https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/Bom for the bom files.
> 
> And pick and place file too...please....

https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/Placement

evilrat, 2021-07-29:
> > > @evilrat Sorry for being late, please check https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/Bom for the bom files.
> > 
> > And pick and place file too...please....
> 
> https://github.com/espressif/esp-drone/tree/master/hardware/ESP32_S2_Drone_V1_2/Placement

Thans bro you are awesome 😎😄

closed 2021-07-29 by evilrat