espressif/esp-drone #16: compile fail

HakFoo, 2021-07-02:
[image]
 region `dram0_0_seg' overflowed by 12640 bytes  
esp32s2
esp-idf 4.3master

leeebo, 2021-07-05:
Seem some components cost much DRAM in the current release/v4.3, you can retry with TAG v4.3
```
git checkout v4.3
git submodule update --init --recursive
```

HakFoo, 2021-07-06:
> Seem some components cost much DRAM in the current release/v4.3, you can retry with TAG v4.3
> 
> ```
> git checkout v4.3
> git submodule update --init --recursive
> ```

it doesn't work

leeebo, 2021-07-06:
Did you modify the code? v4.3 works for me, please make sure checkout success, you can run `git log` to check where you are.

HakFoo, 2021-07-06:
> Did you modify the code? v4.3 works for me, please make sure you successfully checkout, you can run to check where you are.`git log`

just modify the BASE_STACK_SIZE to 1024 in menuconfig and keep task stack size 
[url=https://docs.espressif.com/projects/espressif-esp-drone/en/latest/system.html#sensor-hardware-abstraction]Configure Task Stack Size[\url]

leeebo, 2021-07-07:
> just modify the BASE_STACK_SIZE to 1024 in menuconfig and keep task stack size [...]

[All related tasks stack size](https://github.com/espressif/esp-drone/blob/master/components/config/include/config.h#L178-L214) will be doubled if you modify `BASE_STACK_SIZE` from 512 to 1024 (ESP32-S2 don't have so much RAM). And all task is statically allocated, that's why you got a segment overflow error in compile stage.

HakFoo, 2021-07-09:
> > > Did you modify the code? v4.3 works for me, please make sure you successfully checkout, you can run to check where you are.`git log`
> > 
> > 
> > just modify the BASE_STACK_SIZE to 1024 in menuconfig and keep task stack size
> > [url=https://docs.espressif.com/projects/espressif-esp-drone/en/latest/system.html#sensor-hardware-abstraction]Configure Task Stack Size[\url]
> 
> [All related tasks stack size](https://github.com/espressif/esp-drone/blob/master/components/config/include/config.h#L178-L214) will be doubled if you modify from 512 to 1024 (ESP32-S2 don't have so much RAM). And all task is statically allocated, that's why you got a segment overflow error in compile stage.`BASE_STACK_SIZE`

but I can't find MEM and LOG Task in SYSLOAD with nothing to do.

closed 2022-11-24 by zhanzhaocheng