espressif/esp-drone #17: Please provide source code of the android app

evilrat, 2021-07-25:
https://raw.githubusercontent.com/EspressifApp/EspRelease/master/ESPilot/ESPilot.apk

this app source code please...as esp32 drone don't work on newer app that u provided sourse

evilrat, 2021-07-25:
> https://raw.githubusercontent.com/EspressifApp/EspRelease/master/ESPilot/ESPilot.apk
> 
> this app source code please...as esp32 drone don't work on newer app that u provided sourse

https://github.com/espressif/esp-drone/blob/master/hardware/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf

provide sourse code of android app doesn't work with this version...using esp32

leeebo, 2021-07-29:
@evilrat 
iOS APP Source code: [ESP-Drone-iOS](https://github.com/EspressifApps/ESP-Drone-iOS)
Android APP Source code: [ESP-Drone-Android](https://github.com/EspressifApps/ESP-Drone-Android)

**You can find more details here:**
https://github.com/espressif/esp-drone#introduction

evilrat, 2021-07-29:
> @evilrat
> iOS APP Source code: [ESP-Drone-iOS](https://github.com/EspressifApps/ESP-Drone-iOS)
> Android APP Source code: [ESP-Drone-Android](https://github.com/EspressifApps/ESP-Drone-Android)
> 
> **You can find more details here:**
> https://github.com/espressif/esp-drone#introduction

I tried this app...but it doesnt work for 
https://github.com/espressif/esp-drone/blob/master/hardware/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf
but this app does
https://raw.githubusercontent.com/EspressifApp/EspRelease/master/ESPilot/ESPilot.apk

evilrat, 2021-07-29:
> I tried this app...but it doesnt work for
> https://github.com/espressif/esp-drone/blob/master/hardware/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf
> but this app does [...]

https://github.com/leeebo/ESP-Drone
using this as a firmware

leeebo, 2021-07-29:
To use ESP-Drone code on ESPlane-FC-V1, please connect GPIO14 of ESP32-WROOM-32D to INT pin of MPU6050 by a jumper. You can find detials here: https://docs.espressif.com/projects/espressif-esp-drone/en/latest/hardware.html#configure-target-hardware, not sure if that cause your problem.

P.S. https://github.com/leeebo/ESP-Drone is an Archived project, please use current repo https://github.com/espressif/esp-drone instead.

evilrat, 2021-07-29:
do i need to change pin config if i flash new firmware on 
https://github.com/espressif/esp-drone/blob/master/hardware/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf

leeebo, 2021-07-29:
> do i need to change pin config if i flash new firmware on
> https://github.com/espressif/esp-drone/blob/master/hardware/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf

@evilrat please refer the default gpio config: https://docs.espressif.com/projects/espressif-esp-drone/en/latest/hardware.html#main-board-io-definition-1

evilrat, 2021-07-29:
> > do i need to change pin config if i flash new firmware on
> > https://github.com/espressif/esp-drone/blob/master/hardware/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf
> 
> @evilrat please refer the default gpio config of ESPlane_FC_V1 here: https://docs.espressif.com/projects/espressif-esp-drone/en/latest/hardware.html#main-board-io-definition-1

Thanks bro...🙃😄

closed 2021-07-29 by evilrat