espressif/esp-drone #19: Do you have any future plane to add gps

evilrat, 2021-07-30:
Can we add gps to the drone for back to home and others features......

leeebo, 2021-07-31:
Interesting but redundant, as an air-core rotors mini drone, the flying distance is limited and mostly indoors, where GPS signal is really bad.

evilrat, 2021-07-31:
> Interesting but redundant, as an air-core rotors mini drone, the flying distance is limited and mostly indoors, where GPS signal is really bad. We using [flow ](https://github.com/espressif/esp-drone/tree/master/components/drivers/spi_devices/pmw3901) and [laser](https://github.com/espressif/esp-drone/tree/master/components/drivers/i2c_devices/vl53l1) sensors instead (indoor relatively orientation).
can i add all sensors at one board.....

leeebo, 2021-07-31:
> > can i add all sensors at one board.....

Definitely yes, GPS module like M8N using uart interface, you can connect it with expend IO of esp-drone.