espressif/esp-drone #22: Buying Motherboard

atharvbadkas, 2021-11-07:
Hello,
I'm not much into SMD soldering, Is there any online-store that sells this motherboard ?
Thanks

leeebo, 2021-12-03:
@atharvbadkas Sorry, we can't sell related hardware, nor do we release related firmware, because of the restrictions of the Licence. This project is only for hobbyist learning, but you can make your own [PCB](https://github.com/espressif/esp-drone/tree/master/hardware) and develop it freely (there is no such limit to the source code modification and sharing).

ShubhamBhatt, 2023-10-08:
> @atharvbadkas Sorry, we can't sell related hardware, nor do we release related firmware, because of the restrictions of the Licence. This project is only for hobbyist learning, but you can make your [...]

Hi which PCB company should I use to order?? Please suggest some names. I want it to be fully assembled.