espressif/esp-drone #25: Simple code did not work

yuanyanhui, 2021-12-13:
I cloned the repo, complied and flashed the code without any modifications. Everything was okay. I then added some code to try hovering using the flow deck. Compilation and flashing all went well, but ESP32 kept panic rebooting.

`#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/portmacro.h"
#include "nvs_flash.h"

#include "stm32_legacy.h"
#include "platform.h"
#include "system.h"
#define DEBUG_MODULE "APP_MAIN"
#include "debug_cf.h"
#include "commander.h"

static void positionSet(setpoint_t *setpoint, float x, float y, float z, float yaw)
{
    // memset(setpoint, 0, sizeof(setpoint_t));

    setpoint->mode.x = modeAbs;
    setpoint->mode.y = modeAbs;
    setpoint->mode.z = modeAbs;

    setpoint->position.x = x;
    setpoint->position.y = y;
    setpoint->position.z = z;

    setpoint->mode.yaw = modeAbs;

    setpoint->attitude.yaw = yaw;
}

void app_main()
{
    /*
     * Initialize the platform and Launch the system task
     * app_main will initialize and start everything
     */
    static setpoint_t setpoint;
    /* initialize nvs flash prepare for Wi-Fi */
    esp_err_t ret = nvs_flash_init();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND)
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }

    ESP_ERROR_CHECK(ret);

    /*Initialize the platform.*/
    if (platformInit() == false)
    {
        while (1)
            ; // if  firmware is running on the wrong hardware, Halt
    }

    /*launch the system task */
    systemLaunch();

    while (1)
    {
        vTaskDelay(M2T(10));
        positionSet(&setpoint, 0, 0, 1, 0);
        commanderSetSetpoint(&setpoint, 3);
    }
}`

ELF file SHA256: dbdb58b15d686f3a

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002411a
0x4002411a: esp_restart_noos at E:/ESP32/esp-idf/components/esp32s2/system_api_esp32s2.c:109 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1944
load:0x4004c000,len:0xc9c
load:0x40050000,len:0x2ce0
entry 0x4004c21c
I (24) boot: ESP-IDF v4.3.1-dirty 2nd stage bootloader
I (24) boot: compile time 15:11:17
I (24) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : QIO
I (41) boot.esp32s2: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1c2ech (115436) map        
I (117) esp_image: segment 1: paddr=0002c314 vaddr=3ffc7f60 size=03d04h ( 15620) load      
I (121) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=7f08ch (520332) map       
I (212) esp_image: segment 3: paddr=000af0b4 vaddr=3ffcbc64 size=00c08h (  3080) load      
I (213) esp_image: segment 4: paddr=000afcc4 vaddr=40022000 size=15f58h ( 89944) load      
I (238) esp_image: segment 5: paddr=000c5c24 vaddr=50000000 size=00010h (    16) load      
I (249) boot: Loaded app from partition at offset 0x10000
I (249) boot: Disabling RNG early entropy source...
I (261) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (261) cpu_start: Pro cpu up.
I (274) cpu_start: Pro cpu start user code
I (274) cpu_start: cpu freq: 240000000
I (275) cpu_start: Application information:
I (279) cpu_start: Project name:     ESPDrone
I (284) cpu_start: App version:      cc41b33-dirty
I (290) cpu_start: Compile time:     Dec 13 2021 15:10:16
I (296) cpu_start: ELF file SHA256:  dbdb58b15d686f3a...
I (302) cpu_start: ESP-IDF:          v4.3.1-dirty
I (307) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (321) heap_init: At 3FFDE020 len 0001DFE0 (119 KiB): DRAM
I (327) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (333) spi_flash: detected chip: generic
I (338) spi_flash: flash io: qio
I (345) cpu_start: Starting scheduler on PRO CPU.
I (360) PLATFORM: set active config 
I (360) gpio: GPIO[7]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (367) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (376) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (387) wifi:wifi driver task: 3ffe5c00, prio:23, stack:6656, core=0
I (392) system_api: Base MAC address is not set
I (397) system_api: read default base MAC address from EFUSE
I (407) wifi:wifi firmware version: 88c8747
I (407) wifi:wifi certification version: v7.0
I (411) wifi:config NVS flash: enabled
I (414) wifi:config nano formating: disabled
I (418) wifi:Init data frame dynamic rx buffer num: 32
I (423) wifi:Init management frame dynamic rx buffer num: 32
I (429) wifi:Init management short buffer num: 32
I (433) wifi:Init dynamic tx buffer num: 32
I (437) wifi:Init static rx buffer size: 1600
I (441) wifi:Init static rx buffer num: 10
I (445) wifi:Init dynamic rx buffer num: 32
I (449) wifi_init: rx ba win: 6
I (453) wifi_init: tcpip mbox: 32
I (457) wifi_init: udp mbox: 6
I (460) wifi_init: tcp mbox: 6
I (464) wifi_init: tcp tx win: 5744
I (468) wifi_init: tcp rx win: 5744
I (472) wifi_init: tcp mss: 1440
I (476) wifi_init: WiFi IRAM OP enabled
I (481) wifi_init: WiFi RX IRAM OP enabled
I (979) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
I (1062) wifi:mode : softAP (7c:df:a1:18:ad:2d)
I (1064) wifi:Total power save buffer number: 16
I (1064) wifi:Init max length of beacon: 752/752
I (1064) wifi:Init max length of beacon: 752/752
I (1070) WIFI_UDP: wifi_init_softap complete.SSID:ESP-DRONE_7CDFA118AD2D password:12345678 
I (1078) WIFI_UDP: Socket created
I (1082) WIFI_UDP: Socket bound, port 2390
I (1086) WIFI_UDP: UDP server create socket succeed!!!
assertion "( pxQueue )" failed: file "E:/ESP32/esp-idf/components/freertos/queue.c", line 1656, function: xQueuePeek

abort() was called at PC 0x40019fbe on core 0

Backtrace:0x4002b3ee:0x3ffdf530 0x4002bcc5:0x3ffdf550 0x40033282:0x3ffdf570 0x40019fbe:0x3ffdf5e0 0x4001a42b:0x3ffdf600 0x4002cade:0x3ffdf630 0x400879b4:0x3ffdf670 0x400877f9:0x3ffdf6b0 0x400fedfd:0x3ffdf6d0
0x4002b3ee: panic_abort at E:/ESP32/esp-idf/components/esp_system/panic.c:379

0x4002bcc5: esp_system_abort at E:/ESP32/esp-idf/components/esp_system/system_api.c:112

0x40033282: abort at E:/ESP32/esp-idf/components/newlib/abort.c:46

0x4002cade: xQueuePeek at E:/ESP32/esp-idf/components/freertos/queue.c:1656 (discriminator 
1)

0x400879b4: systemTest at E:\ESP32\esp-drone\build/../components/core/crazyflie/modules/src/system.c:159

0x400877f9: platformParseDeviceTypeString at E:\ESP32\esp-drone\build/../components/platform/platform.c:74

0x400fedfd: heap_bubble_down at E:/ESP32/esp-idf/components/log/log.c:285

yuanyanhui, 2021-12-13:
After adding delay of 3 sec, the system initializes properly, however there is no response from the motors afterwards. The debug message is printed out as expected.

`    /*launch the system task */
    systemLaunch();

    vTaskDelay(M2T(3000));

    while (1)
    {
        vTaskDelay(M2T(10));
        positionSet(&setpoint, 0, 0, 1, 0);
        commanderSetSetpoint(&setpoint, 3);
        DEBUG_PRINT_LOCAL("...");
    }`

yuanyanhui, 2021-12-14:
Checked the outputs (z, vx, vy) of the flow deck which looked quite normal. The drone still could not take off. After initialization, the motors accelerated for 1 second to a high velocity and then stopped.

leeebo, 2021-12-14:
@yuanyanhui Please test with [the esp-drone APP](https://docs.espressif.com/projects/espressif-esp-drone/en/latest/gettingstarted.html?highlight=crazyflie-lib-python#download-and-install-esp-drone-app) for your first flight

yuanyanhui, 2021-12-14:
Tried the phone app. When thrust was increased to about 50%, motors shut off and I had to manually reset to reconnect again. Not sure if this is a hardware problem.

leeebo, 2021-12-14:
@yuanyanhui  The battery voltage needs to be stable when the motor is operating, power lithium batteries are recommended.

yuanyanhui, 2021-12-15:
The battery shouldn't be the problem, as it's fully charged.

yuanyanhui, 2021-12-17:
There is some progress... Somehow thrust control became normal in phone app. Also found out that the Z coordinate axis actually points upward, so to set hovering height, setpoint->position.z should be assigned a negative number. However, the drone still could not take off. At the beginning, all four motors were spinning. After a few seconds, one or two motors would stop. The other two or three motors kept spinning at a high speed.

yuanyanhui, 2021-12-25:
Changed motors and propellers to 720 and 55mm. After tweaking pid and thrust parameters, the drone could hover. Position z was set to a positive number.

closed 2021-12-25 by yuanyanhui

3D21, 2023-10-31:
Can you show the full code of the successful firmware?

carrliitos, 2025-03-15:
@yuanyanhui -- Hello, how were you able to install your PMW3901 and VL53L1X sensors?