espressif/esp-drone #29: VSCode Extension Build Error

IcyFeather233, 2022-01-01:
Environment:
- Linux Mint Latest
- VSCode ESP_IDF extension

I clone the repo and I have set device-target and port to use, then I enter F1 to build the project, and it shows a error:

```
> Executing task: ninja  <

[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable ESPDrone.elf
FAILED: ESPDrone.elf 
: && /home/icyfeather/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++  -mlongcalls   CMakeFiles/ESPDrone.elf.dir/project_elf_src_esp32s2.c.obj  -o ESPDrone.elf  -Wl,--start-group  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/motors/libmotors.a  esp-idf/platform/libplatform.a  esp-idf/i2c_bus/libi2c_bus.a  esp-idf/deck/libdeck.a  esp-idf/mpu6050/libmpu6050.a  esp-idf/ms5611/libms5611.a  esp-idf/hmc5883l/libhmc5883l.a  esp-idf/pmw3901/libpmw3901.a  esp-idf/vl53l1/libvl53l1.a  esp-idf/vl53l0/libvl53l0.a  esp-idf/led/libled.a  esp-idf/eeprom/libeeprom.a  esp-idf/dsp_lib/libdsp_lib.a  esp-idf/wifi/libwifi.a  esp-idf/adc/libadc.a  esp-idf/crazyflie/libcrazyflie.a  esp-idf/buzzer/libbuzzer.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/icyfeather/Projects/espProjects/esp-drone/build/ESPDrone.map  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/console/libconsole.a  esp-idf/json/libjson.a  esp-idf/buzzer/libbuzzer.a  esp-idf/motors/libmotors.a  esp-idf/platform/libplatform.a  esp-idf/i2c_bus/libi2c_bus.a  esp-idf/deck/libdeck.a  esp-idf/mpu6050/libmpu6050.a  esp-idf/ms5611/libms5611.a  esp-idf/hmc5883l/libhmc5883l.a  esp-idf/pmw3901/libpmw3901.a  esp-idf/vl53l1/libvl53l1.a  esp-idf/vl53l0/libvl53l0.a  esp-idf/led/libled.a  esp-idf/eeprom/libeeprom.a  esp-idf/wifi/libwifi.a  esp-idf/adc/libadc.a  esp-idf/crazyflie/libcrazyflie.a  esp-idf/motors/libmotors.a  esp-idf/platform/libplatform.a  esp-idf/i2c_bus/libi2c_bus.a  esp-idf/deck/libdeck.a  esp-idf/mpu6050/libmpu6050.a  esp-idf/ms5611/libms5611.a  esp-idf/hmc5883l/libhmc5883l.a  esp-idf/pmw3901/libpmw3901.a  esp-idf/vl53l1/libvl53l1.a  esp-idf/vl53l0/libvl53l0.a  esp-idf/led/libled.a  esp-idf/eeprom/libeeprom.a  esp-idf/wifi/libwifi.a  esp-idf/adc/libadc.a  esp-idf/crazyflie/libcrazyflie.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/dsp_lib/libdsp_lib.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32s2/libesp32s2.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libcoexist.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libcore.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libespnow.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libmesh.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libnet80211.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libpp.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libsmartconfig.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libwapi.a  /home/icyfeather/esp/esp-idf/components/esp_wifi/lib/esp32s2/libphy.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/app_update/libapp_update.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/xtensa/libxtensa.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/esp32s2/libesp32s2.a  es [... cut ...] .literal.paramInit+0x0): undefined reference to `_param_start'
/home/icyfeather/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-8.4.0/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0xc): undefined reference to `_param_stop'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
The terminal process "bash '-c', 'ninja '" terminated with exit code: 1.
```

closed 2022-01-01 by IcyFeather233

w2016561536, 2022-11-22:
1. Downgrade the idf to 4.3.4
2. Modify the linking script https://docs.espressif.com/projects/espressif-esp-drone/zh_CN/latest/getespidf.html

w2016561536, 2022-11-22:
If you could compile on the latest version of IDF. Please tell me the method, thanks!