espressif/esp-drone #30: doc error fix

IcyFeather233, 2022-01-01:
In Get Started - 2. Install cfclient
if you do as the doc reads, you will meet this error

```
python3 ./bin/cfclient
INFO:cfclient.gui:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2']
INFO:cfclient.utils.input.inputreaders:Successfully initialized [linuxjsdev]
INFO:cfclient.utils.input.inputreaders:Could not initialize [pysdl2]: No SDL2 support on Linux
INFO:cfclient.utils.input.inputinterfaces:Found interfaces: ['leapmotion', 'wiimote', 'zmqpull']
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named 'leapsdk')
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named 'cwiid'
INFO:cfclient.utils.config:Dist config read from /home/icyfeather/crazyflie-clients-python/src/cfclient/configs/config.json
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file
===== 2022.01.01 20:01:53 =====
Traceback (most recent call last):
  File "./bin/cfclient", line 5, in <module>
    main()
  File "/home/icyfeather/crazyflie-clients-python/src/cfclient/gui.py", line 137, in main
    from .ui.main import MainUI
  File "/home/icyfeather/crazyflie-clients-python/src/cfclient/ui/main.py", line 33, in <module>
    import cfclient.ui.tabs
  File "/home/icyfeather/crazyflie-clients-python/src/cfclient/ui/tabs/__init__.py", line 41, in <module>
    from .locopositioning_tab import LocoPositioningTab
  File "/home/icyfeather/crazyflie-clients-python/src/cfclient/ui/tabs/locopositioning_tab.py", line 41, in <module>
    from PyQt5.QtGui import QMessageBox
ImportError: cannot import name 'QMessageBox' from 'PyQt5.QtGui' (/home/icyfeather/myVenv/cfclient/lib/python3.7/site-packages/PyQt5/QtGui.abi3.so)
```

After some search I found [this page](https://forum.bitcraze.io/viewtopic.php?t=4958)

In the end, it says
```
This is now fixed. It was a bug that only recently started causing problem.

I have pushed the fix yesterday to github and release the client 2021.6.1 today. so both "pip3 install cfclient" and "pip3 install -e ." from source should work now.
```

So I think the doc should be change to `pip install cfclient` directly. Or change the git repo source code.

I spend a lot of time to solve the problem and I met others who met the same problem as me. Hope you change the doc in case of more people meet the problem, thx!

leeebo, 2022-01-13:
@IcyFeather233 Thanks for your report, will fix it soon

leeebo, 2024-08-14:
@IcyFeather233  The bug has already fixed, please update the latest code and install the new PC client refer https://github.com/qljz1993/crazyflie-clients-python.git

closed 2024-08-14 by leeebo