espressif/esp-drone #31: ESP drone linking issue

KC-coder, 2022-01-12:
**I am working with ESP drone code** 

**I have't made any changes to it yet just traying to compile it by i am getting the linking issue** 
'
my IDF version : 4.4
my OS : Windows 10 

THIS is the error i am getting 
```
[1166/1208] Building C object esp-idf/crazyflie/CMakeFiles/__idf_crazyflie.dir/modules/src/pptraj.c.obj
In file included from ../components/core/crazyflie/modules/interface/pptraj.h:41,
                 from ../components/core/crazyflie/modules/src/pptraj.c:40:
../components/core/crazyflie/modules/interface/math3d.h: In function 'fcloseulps':
../components/core/crazyflie/modules/interface/math3d.h:65:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  int ia = *((int *)&a);
            ~^~~~~~~~~~
../components/core/crazyflie/modules/interface/math3d.h:66:13: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
  int ib = *((int *)&b);
            ~^~~~~~~~~~
[1206/1208] Linking CXX executable ESPDrone.elf
FAILED: ESPDrone.elf 
cmd.exe /C "cd . && D:\ESP_Work\tool\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\ESPDrone.elf.rsp -o ESPDrone.elf  && cd ."
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: missing --end-group; added as last command line option
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(log.c.obj):(.literal.logInit+0x0): undefined reference to `_log_start'
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(log.c.obj):(.literal.logInit+0x4): undefined reference to `_log_stop'
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0x0): undefined reference to `_param_start'
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0xc): undefined reference to `_param_stop'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

And this is my c_cpp_properties.json file settings 

```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "d:\\ESP_Work\\tool\\tools\\xtensa-esp32-elf\\esp-2021r2-8.4.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
```

and this is my  CMakeLists file

```
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

set(__PROJECT_GROUP_LINK_COMPONENTS ON)

set(PLANE_COMPONENT_DIRS ${PLANE_COMPONENT_DIRS}
                        "./components/core"
                        "./components/drivers"
                        "./components/drivers/general"
                        "./components/drivers/i2c_devices"
                        "./components/drivers/spi_devices"
                         "./components/lib" )

set(EXTRA_COMPONENT_DIRS "${EXTRA_COMPONENT_DIRS} ${PLANE_COMPONENT_DIRS}")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(ESPDrone)

```

I have traied many things but same error persists

any help would be appreciated 
thank you.

ou2356, 2022-02-19:
You should edit its link file as this link(https://docs.espressif.com/projects/espressif-esp-drone/zh_CN/latest/getespidf.html#esp32-esp32-s2). Then you can solve four errors.
`d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(log.c.obj):(.literal.logInit+0x0): undefined reference to `_log_start'
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(log.c.obj):(.literal.logInit+0x4): undefined reference to `_log_stop'
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0x0): undefined reference to `_param_start'
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/crazyflie/libcrazyflie.a(param.c.obj):(.literal.paramInit+0xc): undefined reference to `_param_stop'`

ou2356, 2022-02-19:
But I don't know how to deal with this problem.
```
d:/esp_work/tool/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: missing --end-group; added as last command line option
```

ast626, 2022-09-15:
3Q，using esp-idfv4.4，link files path is : {IDF_PATH}/components/esp_system\ld\esp32s2\sections.ld.in，
build complete
ESPDrone.bin binary size 0xb8460 bytes. Smallest app partition is 0x100000 bytes. 0x47ba0 bytes (28%) free.

closed 2022-11-24 by zhanzhaocheng