espressif/esp-drone #32: I am traying to use the code

KC-coder, 2022-01-15:
I cloned the code but every time when i try to run the code it got stuck here.
i tried to use the mobile app (android version , both old and new version) to control the motors but its not working, their is no response even though mobile is getting connected to ESP and also APP is getting connected 

Here's what i am getting on the monitor :

`ad.
  if StrictVersion(serial.VERSION) < StrictVersion('3.3.0'):
--- idf_monitor on \\.\COM3 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

;32mI (442) heap_init: At 3FFC9E80 len 00016180 (88 KiB): DRAMts Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7436 
ho 0 tail 12 room 4      
load:0x40078000,len:13904
load:0x40080400,len:5296
0x40080400: _init at ??:?

entry 0x40080710
I (28) boot: ESP-IDF v4.2.2-dirty 2nd stage bootloader
I (28) boot: compile time 19:09:01
I (28) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) qio_mode: Enabling default flash chip QIO
I (44) boot.esp32: SPI Speed      : 80MHz
I (49) boot.esp32: SPI Mode       : QIO
I (53) boot.esp32: SPI Flash Size : 4MB
I (58) boot: Enabling RNG early entropy source...
I (63) boot: Partition Table:
I (67) boot: ## Label            Usage          Type ST Offset   Length
I (74) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (81) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (89) boot:  2 factory          factory app      00 00 00010000 00100000
I (96) boot: End of partition table
I (101) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f608 (128520) map
I (155) esp_image: segment 1: paddr=0x0002f630 vaddr=0x3ffb0000 size=0x009e8 (  2536) load
I (156) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x891b0 (561584) map
0x400d0020: _stext at ??:?

I (327) esp_image: segment 3: paddr=0x000b91d8 vaddr=0x3ffb09e8 size=0x03614 ( 13844) load
I (331) esp_image: segment 4: paddr=0x000bc7f4 vaddr=0x40080000 size=0x182d0 ( 99024) load
0x40080000: _WindowOverflow4 at D:/ESP_Work/IDF/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (381) boot: Loaded app from partition at offset 0x10000
I (381) boot: Disabling RNG early entropy source...
I (382) cpu_start: Pro cpu up.
I (385) cpu_start: Application information:
I (390) cpu_start: Project name:     ESPDrone
I (395) cpu_start: App version:      cc41b33-dirty
I (401) cpu_start: Compile time:     Jan 15 2022 19:08:43
I (407) cpu_start: ELF file SHA256:  0061931a35e9902f...
I (413) cpu_start: ESP-IDF:          v4.2.2-dirty
I (418) cpu_start: Starting app cpu, entry point is 0x40081ac8
0x40081ac8: esp_cache_err_get_cpuid at D:/ESP_Work/IDF/esp-idf/components/esp32/cache_err_int.c:84

I (0) cpu_start: App cpu up.
I (429) heap_init: Initializing. RAM available for dynamic allocation:
I (436) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (442) heap_init: At 3FFC9E80 len 00016180 (88 KiB): DRAM
I (448) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (454) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (461) heap_init: At 400982D0 len 00007D30 (31 KiB): IRAM
I (467) cpu_start: Pro cpu start user code
I (484) spi_flash: detected chip: generic
I (484) spi_flash: flash io: qio
I (484) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (531) PLATFORM: set active config 
I (532) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (534) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (543) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (555) wifi:wifi driver task: 3ffd16cc, prio:23, stack:6656, core=0
I (559) system_api: Base MAC address is not set
I (564) system_api: read default base MAC address from EFUSE
I (581) wifi:wifi firmware version: bb6888c
I (581) wifi:wifi certification version: v7.0
I (581) wifi:config NVS flash: enabled
I (581) wifi:config nano formating: disabled
I (586) wifi:Init data frame dynamic rx buffer num: 32
I (590) wifi:Init management frame dynamic rx buffer num: 32
I (596) wifi:Init management short buffer num: 32
I (600) wifi:Init dynamic tx buffer num: 32
I (604) wifi:Init static rx buffer size: 1600
I (608) wifi:Init static rx buffer num: 10
I (612) wifi:Init dynamic rx buffer num: 32
I (616) wifi_init: rx ba win: 6
I (620) wifi_init: tcpip mbox: 32
I (624) wifi_init: udp mbox: 6
I (627) wifi_init: tcp mbox: 6
I (631) wifi_init: tcp tx win: 5744
I (635) wifi_init: tcp rx win: 5744
I (639) wifi_init: tcp mss: 1440
I (643) wifi_init: WiFi IRAM OP enabled
I (648) wifi_init: WiFi RX IRAM OP enabled
I (1135) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1217) wifi:mode : softAP (ec:94:cb:5c:9e:49)
I (1218) wifi:Total power save buffer number: 16
I (1219) wifi:Init max length of beacon: 752/752
I (1219) wifi:Init max length of beacon: 752/752
I (1224) WIFI_UDP: wifi_init_softap complete.SSID:ESP-DRONE_EC94CB5C9E49 password:12345678
I (1233) WIFI_UDP: Socket created
I (1236) WIFI_UDP: Socket bound, port 2390
I (1241) WIFI_UDP: UDP server create socket succeed!!!
I (1747) SYS: ----------------------------

I (1747) SYS: ESPlane 2.0  is up and running!

I (1747) I2CDRV:  i2c 1 driver install return = 0
I (1751) EEPROM: spi_flash_init ... !
I (1756) EEPROM: spi_flash_read ok !
eFuse Two Point: NOT supported
eFuse Vref: Supported
Characterized using eFuse Vref
I (2523) I2CDRV:  i2c 0 driver install return = 0
I (2523) SENSORS: MPU6050 I2C connection [OK].

W (2878) ZR2: Z-down sensor [FAIL]

W (2878) SENSORS: VL53L1X I2C connection [FAIL].

I (2878) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
W (2933) SENSORS: PMW3901 SPI connection [FAIL].

I (2933) SENSORS: sensors init done
I (2933) SENSORS: pitch_calib = 0.000000,roll_calib = 0.000000
I (2938) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (2948) ESTIMATOR: Using Complementary (1) estimator

I (2954) POSITION_CONTROLLER: thrustBase = 36000,thrustMin  = 20000
I (2961) SYS: wifilinkTest = 1 
I (3464) SYS: pmTest = 1
I (3464) SYS: workerTest = 1
I (3464) SYS: systemTest = 1 
I (3464) SYS: configblockTest = 1 
I (3467) COMM: wifilinkTest = 1 
I (3471) COMM: crtpTest = 1 
I (3474) COMM: crtpserviceTest = 1 
I (3479) COMM: platformserviceTest = 1 
I (3483) COMM: consoleTest = 1 
I (3487) COMM: paramTest = 1 
I (3491) SYS: commTest = 1 
I (3494) SYS: commanderTest = 1 
I (5313) SYS: stabilizerTest = 1 
I (5313) SYS: estimatorKalmanTaskTest = 1 
I (5313) SYS: memTest = 1 
I (5314) STAB: Wait for sensor calibration...

I (5319) STAB: Ready to fly.

I (5321) SYS: systemStart ! selftestPassed = 1
i am in workerLoop while1 
i go to sensorTask 
i go to sensorTask's while loop
I (6749) SYSLOAD: Task dump

I (6749) SYSLOAD: Load  Stack left      Name    PRI

I (6749) SYSLOAD: 0.15  1632    Tmr Svc         1

I (6752) SYSLOAD: 97.86         992     IDLE1   0

I (6756) SYSLOAD: 86.73         960     IDLE0   0

I (6761) SYSLOAD: 0.01  5612    CRTP-RX         2

I (6766) SYSLOAD: 0.21  3580    PWRMGNT         0

I (6770) SYSLOAD: 0.04  2720    tiT     18

I (6775) SYSLOAD: 0.27  1548    UDP_TX  3

I (6780) SYSLOAD: 0.00  3308    UDP_RX  3

I (6784) SYSLOAD: 0.00  544     WIFILINK        3

I (6789) SYSLOAD: 0.49  512     ipc1    24

I (6794) SYSLOAD: 0.00  2560    LOG     1

I (6798) SYSLOAD: 0.00  1556    PARAM   1

I (6803) SYSLOAD: 0.00  1508    CMDHL   2

I (6807) SYSLOAD: 0.00  1572    CRTP-TX         2

I (6812) SYSLOAD: 0.08  516     ipc0    2

I (6816) SYSLOAD: 0.00  1600    KALMAN  2

I (6821) SYSLOAD: 0.14  3160    STABILIZER      5

I (6826) SYSLOAD: 0.00  1584    LEDSEQCMD       1

I (6831) SYSLOAD: 3.28  3828    SYSTEM  2

I (6836) SYSLOAD: 0.05  4444    SENSORS         4

I (6840) SYSLOAD: 0.06  3412    esp_timer       22

I (6846) SYSLOAD: 10.17         4520    wifi    23

I (6850) SYSLOAD: 0.01  1920    sys_evt         20

I (6855) SYSLOAD: Free heap: 143864 bytes

`
Hardware used:
ESP32 WROOM 32
also tried on ESP32 WROOM - 32E
Sensors connected : MPU6050 ( connected no other sensor apart from this)

Version's:
IDF Version : 4.2.2
VS code IDF extension version: 1.2

Tried to increase Stack size : from 1024 to 2048 but had same issue 
tried to android App version 1 and 2 not had same issue 
Even tried to run the code by commenting out the sensor calibration function 

Can anyone please help. I don't know where to start debugging

leeebo, 2022-03-24:
@KC-coder Can't find any issues from your log, it seems `Ready to fly`. 

Please check if you `connect` to the drone before any operator

[image]

MrityunjaiKumar, 2022-06-23:
Still after that it not working, On connect esp try to send some UDP packet but it doesn't reflect in cfcleint or app.

Here are the  logs
`(1200) wifi:Init max length of beacon: 752/752
I (1204) WIFI_UDP: wifi_init_softap complete.SSID:ESP-DRONE_58BF25222FF5 password:12345678
I (1213) WIFI_UDP: Socket created
I (1217) WIFI_UDP: Socket bound, port 2390
I (1222) WIFI_UDP: UDP server create socket succeed!!!
I (1728) SYS: ----------------------------

I (1728) SYS: ESPlane 2.0  is up and running!

I (1728) I2CDRV:  i2c 1 driver install return = 0
I (1732) EEPROM: spi_flash_init ... !
I (1736) EEPROM: spi_flash_read ok !
eFuse Two Point: NOT supported
eFuse Vref: Supported
Characterized using eFuse Vref
I (2503) I2CDRV:  i2c 0 driver install return = 0
I (2503) SENSORS: MPU6050 I2C connection [OK].

I (2856) SENSORS: sensors init done
I (2856) SENSORS: pitch_calib = 0.000000,roll_calib = 0.000000
I (2856) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1 
I (2865) ESTIMATOR: Using Complementary (1) estimator

I (2871) POSITION_CONTROLLER: thrustBase = 36000,thrustMin  = 20000
I (2878) SYS: wifilinkTest = 1 
I (3381) SYS: pmTest = 1
I (3381) SYS: workerTest = 1
I (3381) SYS: systemTest = 1 
I (3381) SYS: configblockTest = 1 
I (3384) COMM: wifilinkTest = 1 
I (3388) COMM: crtpTest = 1 
I (3391) COMM: crtpserviceTest = 1 
I (3396) COMM: platformserviceTest = 1 
I (3400) COMM: consoleTest = 1 
I (3404) COMM: paramTest = 1 
I (3408) SYS: commTest = 1 
I (3411) SYS: commanderTest = 1 
I (6730) SYSLOAD: Task dump

I (6730) SYSLOAD: Load  Stack left      Name    PRI

I (6731) SYSLOAD: 0.15  1632    Tmr Svc         1

I (6733) SYSLOAD: 98.18         992     IDLE1   0

I (6737) SYSLOAD: 87.15         992     IDLE0   0

I (6742) SYSLOAD: 0.01  5604    CRTP-RX         2

I (6747) SYSLOAD: 0.04  2720    tiT     18

I (6751) SYSLOAD: 3.08  3836    SYSTEM  2

I (6756) SYSLOAD: 0.26  1524    UDP_TX  3

I (6761) SYSLOAD: 0.00  3300    UDP_RX  3

I (6765) SYSLOAD: 0.00  552     WIFILINK        3

I (6770) SYSLOAD: 0.39  512     ipc1    24

I (6775) SYSLOAD: 0.00  1592    LEDSEQCMD       1

I (6779) SYSLOAD: 0.00  3620    PWRMGNT         0

I (6784) SYSLOAD: 0.00  2568    LOG     1

I (6789) SYSLOAD: 0.00  1564    PARAM   1

I (6793) SYSLOAD: 0.00  1516    CMDHL   2

I (6798) SYSLOAD: 0.00  1608    KALMAN  2

I (6803) SYSLOAD: 0.00  1548    CRTP-TX         2

I (6807) SYSLOAD: 0.08  516     ipc0    2

I (6812) SYSLOAD: 0.00  4548    SENSORS         4

I (6817) SYSLOAD: 0.00  4528    STABILIZER      5

I (6822) SYSLOAD: 0.06  3412    esp_timer       22

I (6827) SYSLOAD: 10.21         4496    wifi    23

I (6831) SYSLOAD: 0.01  1924    sys_evt         20

I (6836) SYSLOAD: Free heap: 146604 bytes

I (173287) wifi:new:<13,2>, old:<13,2>, ap:<13,2>, sta:<255,255>, prof:13
I (173288) wifi:station: c0:b8:83:dd:45:3c join, AID=1, bgn, 40D
I (173308) WIFI_UDP: station c0:b8:83:dd:45:3c join, AID=1
I (173325) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.43.43
I (208989) WIFI_UDP: Send data to
I (208989) WIFI_UDP:  data_send[0] = FD 
I (208989) WIFI_UDP:  data_send[1] = 42 
I (208991) WIFI_UDP:  data_send[2] = 69 
I (208995) WIFI_UDP:  data_send[3] = 74 
I (209000) WIFI_UDP:  data_send[4] = 63 
I (209005) WIFI_UDP:  data_send[5] = 72 
I (209009) WIFI_UDP:  data_send[6] = 61 
I (209014) WIFI_UDP:  data_send[7] = 7A 
I (209018) WIFI_UDP:  data_send[8] = 65 
I (209023) WIFI_UDP:  data_send[9] = 20 
I (209028) WIFI_UDP:  data_send[10] = 43 
I (209032) WIFI_UDP:  data_send[11] = 72 
I (209037) WIFI_UDP:  data_send[12] = 61 
I (209042) WIFI_UDP:  data_send[13] = 7A 
I (209046) WIFI_UDP:  data_send[14] = 79 
I (209051) WIFI_UDP:  data_send[15] = 66 
I (209056) WIFI_UDP:  data_send[16] = 6C 
I (209060) WIFI_UDP:  data_send[17] = 69 
I (209065) WIFI_UDP:  data_send[18] = 65 
I (209070) WIFI_UDP:  data_send[19] = 00 
I (209074) WIFI_UDP:  data_send[20] = 00 
I (209079) WIFI_UDP:  data_send[21] = 00 
I (209084) WIFI_UDP:  data_send[22] = 00 
I (209089) WIFI_UDP:  data_send[23] = 00 
I (209093) WIFI_UDP:  data_send[24] = 00 
I (209098) WIFI_UDP:  data_send[25] = 00 
I (209103) WIFI_UDP:  data_send[26] = 00 
I (209107) WIFI_UDP:  data_send[27] = 00 
I (209112) WIFI_UDP:  data_send[28] = 00 
I (209117) WIFI_UDP:  data_send[29] = 00 
I (209121) WIFI_UDP:  data_send[30] = 00 
I (209126) WIFI_UDP:  data_send[31] = FA 
I (209893) WIFI_UDP: Send data to
I (209893) WIFI_UDP:  data_send[0] = DD 
I (209893) WIFI_UDP:  data_send[1] = 00 
I (209895) WIFI_UDP:  data_send[2] = 04 
I (209900) WIFI_UDP:  data_send[3] = 00 
I (209904) WIFI_UDP:  data_send[4] = 00 
I (209909) WIFI_UDP:  data_send[5] = 00 
I (209914) WIFI_UDP:  data_send[6] = E1 
I (209919) WIFI_UDP: Send data to
I (209922) WIFI_UDP:  data_send[0] = 5D 
I (209927) WIFI_UDP:  data_send[1] = 05 
I (209931) WIFI_UDP:  data_send[2] = 00 
I (209936) WIFI_UDP:  data_send[3] = 00 
I (209941) WIFI_UDP:  data_send[4] = 62 
I (209946) WIFI_UDP: Send data to
I (209949) WIFI_UDP:  data_send[0] = 50 
I (209954) WIFI_UDP:  data_send[1] = 03 
I (209958) WIFI_UDP:  data_send[2] = 22 
I (209963) WIFI_UDP:  data_send[3] = 01 
I (209968) WIFI_UDP:  data_send[4] = 6C 
I (209972) WIFI_UDP:  data_send[5] = F5 
I (209977) WIFI_UDP:  data_send[6] = 04 
I (209981) WIFI_UDP:  data_send[7] = 0B 
I (209986) WIFI_UDP:  data_send[8] = 10 
I (209991) WIFI_UDP:  data_send[9] = 80 
I (209995) WIFI_UDP:  data_send[10] = 76 `

Not able to figure out the issue

closed 2023-07-28 by zhanzhaocheng

leeebo, 2024-08-13:
@KC-coder @pedrominatel The bug has already fixed, please update the latest code and install the new PC client refer https://github.com/qljz1993/crazyflie-clients-python.git