espressif/esp-drone #38: esp-drone failed to connect to cfclient

w2016561536, 2022-08-11:
I have tried the cfclient and cflib from crazyflie official and modified by espressif.

-------------------------------------------------------------------------------------------

INFO:cfclient.gui:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2']
INFO:cfclient.utils.input.inputreaders:Successfully initialized [linuxjsdev]
INFO:cfclient.utils.input.inputreaders:Could not initialize [pysdl2]: No SDL2 support on Linux
INFO:cfclient.utils.input.inputinterfaces:Found interfaces: ['leapmotion', 'wiimote', 'zmqpull']
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named 'leapsdk')
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named 'cwiid'
INFO:cfclient.utils.config:Dist config read from /home/wyt/cf-client/crazyflie-clients-python/src/cfclient/configs/config.json
INFO:cfclient.utils.config:Config file read from [/home/wyt/.config/cfclient/config.json]
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file
QSocketNotifier: Can only be used with threads started with QThread
INFO:cfclient.utils.zmq_param:Biding ZMQ for parameters at tcp://*:1213
INFO:cfclient.utils.zmq_led_driver:Biding ZMQ for LED driverat tcp://*:1214
INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)]
INFO:cfclient.ui.main:Auto reconnect enabled: True
INFO:cfclient.ui.tabs.QualisysTab:Switching Flight Mode to: FlightModeStates.DISCONNECTED
INFO:cflib.crazyflie:Callback->Connection initialized[udp://192.168.43.42]
INFO:cflib.crazyflie:We are connected[udp://192.168.43.42], request connection setup
INFO:cflib.crazyflie:Callback->Connected to [udp://192.168.43.42]
INFO:cflib.crazyflie.platformservice:Procotol version: 4
INFO:cflib.crazyflie.toc:TOC for port [5] found in cache
INFO:cflib.crazyflie:Log TOC finished updating

--------------------------------------------------------------------------

And the ui stuck at " Connecting to udp:xxxxxxx "

However I have tried to rollback to '4869b0342b4058f8a631f94c8a5e83735625e670' . And it can successfully connect.

Heintsing, 2022-08-20:
I have the same question, how to solve this problem?

w2016561536, 2022-08-20:
> I have the same question, how to solve this problem?

rollback to 4869b0342b4058f8a631f94c8a5e83735625e670

hellvesper, 2023-07-26:
https://github.com/espressif/esp-drone/blob/3e8ffe378c8fc482a5b7a0654804159309f94ddb/components/core/crazyflie/modules/src/system.c#L203C15-L203C15

commented `memInit()` blocks connection in `cfclient`, as it awaits answer from this Port

leeebo, 2023-07-27:
@hellvesper Memory module is not fully implemented, so I just comment it out. If comment this line can fix, would you please submit a Pull Request here

hellvesper, 2023-07-27:
> @hellvesper Memory module is not fully implemented, so I just comment it out. If comment this line can fix, would you please submit a Pull Request here

I see. I had no any issues with active memory task, I created a PR https://github.com/espressif/esp-drone/pull/53 with uncommented line

closed 2023-07-28 by zhanzhaocheng

ViniciusAbrao, 2023-09-22:
> https://github.com/espressif/esp-drone/blob/3e8ffe378c8fc482a5b7a0654804159309f94ddb/components/core/crazyflie/modules/src/system.c#L203C15-L203C15
> 
> commented `memInit()` blocks connection in `cfclient`, as it awaits answer from this Port

Please, I have tried this and it still does not work. Any idea?

hellvesper, 2023-09-24:
> > https://github.com/espressif/esp-drone/blob/3e8ffe378c8fc482a5b7a0654804159309f94ddb/components/core/crazyflie/modules/src/system.c#L203C15-L203C15
> > commented `memInit()` blocks connection in `cfclient`, as it awaits answer from this Port
> 
> Please, I have tried this and it still does not work. Any idea? @leeebo @hellvesper

You should uncomment it