espressif/esp-drone #40: How to Upgrade to ESP32-S3

sd196821, 2022-09-11:
I'd like to use ESP32-S3 core instead of original S2,  besides editing _"main\Kconfig.projbuild"_ and _"sdkconfig"_ files, could anyone plz give me some help or instructions ?

w2016561536, 2022-11-22:
Excuse me, I found that the project can't be compiled correctly on the IDF version above 4.4. However, esp32-s3 is preview state in the IDF 4.3.4. How do you solve it?

sd196821, 2022-11-25:
> Excuse me, I found that the project can't be compiled correctly on the IDF version above 4.4. However, esp32-s3 is preview state in the IDF 4.3.4. How do you solve it?

My s3 drone works well in v4.4 IDF

leeebo, 2022-11-25:
@sd196821 @w2016561536  You can now build with IDF4.4 or 5.0 (no need to modify link scrip), please refer 5e88fcec676753448f8f66262af4352ca463a4bc.

sd196821, 2022-11-25:
> @sd196821 @w2016561536 You can now build with IDF4.4 or 5.0 (no need to modify link scrip), please refer [5e88fce](https://github.com/espressif/esp-drone/commit/5e88fcec676753448f8f66262af4352ca463a4bc).

WOW, this project updated yesterday, NICE WORK, although I've managed to fly with s3 nearly 2 months ago.
By the way, are cfclient GCS ( cfclient PC/Phone ) supporting data transportation with drone now?