espressif/esp-drone #49: 一直重启，提示这个错误是怎么回事

tiandajiang1, 2023-03-06:
在网上看是加大什么stack，加大了之后也是这也，这是咋回事

[0;32mI (3407) SYS: commTest = 1 [0m
[0;32mI (3410) SYS: commanderTest = 1 [0m
[0;32mI (5229) SYS: stabilizerTest = 1 [0m
[0;32mI (5229) SYS: estimatorKalmanTaskTest = 1 [0m
[0;32mI (5229) SYS: memTest = 1 [0m
[0;32mI (5230) STAB: Wait for sensor calibration...
[0m
[0;32mI (5235) SYS: systemStart ! selftestPassed = 1[0m
[0;32mI (6169) SYSLOAD: Free
***ERROR*** A stack overflow in task Tmr Svc has been detected.

Backtrace: 0x40375eb6:0x3fcf5070 0x4037dc99:0x3fcf5090 0x403819e6:0x3fcf50b0 0x4037fae4:0x3fcf5120 0x4037dd50:0x3fcf5140 0x4037dd46:0x00000000 |<-CORRUPTED

ELF file SHA256: e060ce0b8b2e73a8

Rebooting...

leeebo, 2023-03-07:
哪个版本 ESP-IDF?

leeebo, 2023-03-07:
`idf.py menuconfig` `Component config -> FreeRTOS -> FreeRTOS timer task stack size` 改大，先尝试改为 2 倍。不同 IDF 版本该项配置的位置可能不同，你找一下

tiandajiang1, 2023-03-07:
解决了，把print输出屏蔽就好了，现在又有个问题，连接PC cfclient 没反应

tiandajiang1, 2023-03-07:
https://github.com/qljz1993/crazyflie-clients-python.git是这个把？
安卓可以连接，PC一直没没有，提示cfclient.utils.input.inputreaders.pysdl2:Found 0 devices