espressif/esp-drone #54: 請教 ESP-Drone 的 cfclient 無法啟動

cls3free, 2023-08-07:
电腦 win10 已經 安裝 python 3.11 , ubuntu22.04 , PyQt5 5.15.9 , Git , 和 所有必要依賴
但 cmd 或是 ubuntu 輸入指令 cfclient or python3 ./bin/cfclient
或是直接 在exe.win-amd64-3.11 文件夾中 開啟cfclient.exe
也是沒反應 or error
請問怎麼解決? :|  :?: 

[image]
[image]
[image]
[image]
[image]

leeebo, 2023-08-23:
@cls3free Hi，请参考 [gettingstarted.html#cfclient](https://docs.espressif.com/projects/espressif-esp-drone/zh_CN/latest/gettingstarted.html#cfclient) 在 PowerShell 中运行指令，分别安装 cflib 和 cfclinet，然后直接运行 cfclient

leeebo, 2023-08-23:
如果遇到 `CRITICAL:cfclient.gui:No pysdl2 installation found, exiting!`, 请参考 https://github.com/bitcraze/crazyflie-clients-python/issues/259

loggera, 2024-03-21:
window上如何安装？

leeebo, 2024-08-13:
@loggera @cls3free The bug has already fixed, please update the latest code and install the new PC client refer https://github.com/qljz1993/crazyflie-clients-python.git

新版本 Linux 和 Windows 安装过程一致