espressif/esp-drone #57: ESP32-Drone CFClient do not start

acbramos, 2023-11-27:
This error avoid that cfclient to start:
$ python3 ./bin/cfclient
Traceback (most recent call last):
  File "/home/ramos/crazyflie-clients-python/./bin/cfclient", line 2, in <module>
    from cfclient.gui import main
  File "/home/ramos/crazyflie-clients-python/src/cfclient/gui.py", line 37, in <module>
    from quamash import QSelectorEventLoop
  File "/home/ramos/.local/lib/python3.10/site-packages/quamash/__init__.py", line 594, in <module>
    from ._unix import _SelectorEventLoop
  File "/home/ramos/.local/lib/python3.10/site-packages/quamash/_unix.py", line 44, in <module>
    class _SelectorMapping(collections.Mapping):
AttributeError: module 'collections' has no attribute 'Mapping'

lovefool, 2024-07-31:
Have you found any solution yet?

lovefool, 2024-08-01:
I could start the latest version of cfcllient from Bitcraze, but can't connect to my ESP-Drone.
Uninstall several old packages using "pip uninstall xxxxxx" , then "pip install cfclient" to install the latest version.
I hope your success.

closed 2024-08-13 by zhanzhaocheng

leeebo, 2024-08-13:
@lovefool @acbramos The bug has already fixed, please update the latest code and install the new PC client refer https://github.com/qljz1993/crazyflie-clients-python.git