espressif/esp-drone #61: how to make this work on icm 20948

NBL-NZ, 2024-04-09:
i want to make drone with esp32 and icm 20948. How to change the firmware to make it work with icm 20948

mvschaefer, 2024-04-09:
@NBL-NZ - i may be able to help modify the firmware for the 20948.  have you already built a version of the schematic using the 20948?

NBL-NZ, 2024-04-11:
@mvschaefer yes  i have built using 20948.

mvschaefer, 2024-04-11:
@NBL-NZ - can you share your schematic?  specifically, are you using SPI or I2C, and is AD0 tied to GND (I2C address --> 0x68) or VDD (I2C address --> 0x69)?

mvschaefer, 2024-04-12:
then the default address should be 0x69

a good starting point for src code would be to replace the components/drivers/i2c_devices/mpu6050/ folder w/a local/modified version of:
- https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary (if you are using arduino this could be easy?..)
OR
- https://github.com/stephendpmurphy/icm20948

if you are "past this point" already and things arent working, access to src code will likely be required...

NBL-NZ, 2024-05-27:
> then the default address should be 0x69
>
> a good starting point for src code would be to replace the components/drivers/i2c_devices/mpu6050/ folder w/a local/modified version of:
>
> * [...]

Is there is any option to controll flight mode through mobile application  .currently i have set pressure sensor to this drone. but didnt see any option regarding flight mode for using altitude control  mode .i have checked complete app setting there is no option to switch or use flight modes using expansion module.. Is there is somthing i missing ?