espressif/esp-drone #62: stabilizerTask是如何以1000Hz的频率执行的呢？

HelloIDTC, 2024-04-29:
在stabilizerTask任务的while(1)循环中并没有调用vTaskDelayUntil之类的时间函数，那么stabilizerTask是如何以1000Hz的频率执行的呢？

leeebo, 2024-04-29:
@HelloIDTC `stabilizerTask` 是传感器事件驱动的，传感器是 1000 Hz 报中断，所以 `stabilizerTask` 以同样的频率执行

HelloIDTC, 2024-04-29:
是不是执行到这里 sensorsWaitDataReady();然后stabilizerTask任务阻塞，等待传感器1ms中断然后唤醒stabilizerTask任务？
那么传感器1000Hz的中断时间是非常精确的么？会不会有时候有抖动？
我看crazyflie的源码原来在stabilizerTask中会调用vTaskDelayUntil(&lastWakeTime, F2T(RATE_MAIN_LOOP));来延时，后来也改成等待传感器中断的方式了，不知道有什么优势？

HelloIDTC, 2024-04-29:
@leeebo 感谢回复
是不是执行到sensorsWaitDataReady()这里，然后stabilizerTask任务阻塞，等待传感器1ms中断然后唤醒stabilizerTask任务？
那么传感器1000Hz的中断时间是非常精确的么？会不会有时候有抖动？
我看crazyflie的源码原来在stabilizerTask中会调用vTaskDelayUntil(&lastWakeTime, F2T(RATE_MAIN_LOOP));来延时，后来也改成等待传感器中断的方式了，不知道有什么优势

leeebo, 2024-04-29:
@HelloIDTC 传感器来的中断信号是基本准确的，你可以用逻辑分析仪抓一下

HelloIDTC, 2024-04-30:
嗯嗯，明白了，感谢，中断信号的时间应该是和数据输出速率（ODR寄存器）的配置是一致的？

HelloIDTC, 2024-04-30:
@leeebo 嗯嗯，明白了，感谢，中断信号的时间应该是和数据输出速率（ODR寄存器）的配置是一致的？

leeebo, 2024-04-30:
> @leeebo 嗯嗯，明白了，感谢，中断信号的时间应该是和数据输出速率（ODR寄存器）的配置是一致的？

Yes

HelloIDTC, 2024-04-30:
@leeebo 好的，感谢感谢

closed 2024-04-30 by HelloIDTC