espressif/esp-drone #66: How to connect my ESP32-S2-Drone v1.2 to cfclient?

lovefool, 2024-07-30:
My issue is maybe same as #33 that is already closed. (https://github.com/espressif/esp-drone/issues/33)

I have installed the latest version of cfclient frim Bitcraze, and possible to run on my Windows11.
But I don't know how to cionnect cfclient to ESP32-S2-Drone v1.2.

`Z:\ESP-Drone\crazyflie-clients-python>python ./bin/cfclient
UserWarning: Using SDL2 binaries from pysdl2-dll 2.24.0
INFO:cfclient.gui:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2']
INFO:cfclient.utils.input.inputreaders:Could not initialize [linuxjsdev]: Only supported on Linux
INFO:cfclient.utils.input.inputreaders:Successfully initialized [pysdl2]
INFO:cfclient.utils.input.inputinterfaces:Found interfaces: ['leapmotion', 'wiimote', 'zmqpull']
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named 'leapsdk')
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named 'cwiid'
INFO:cfclient.utils.config:Dist config read from z:\esp-drone\crazyflie-clients-python\src\cfclient/configs/config.json
INFO:cfclient.utils.config:Config file read from [C:\Users\jayte.HP-PAVILION\AppData\Local\Bitcraze\cfclient/config.json]
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file
INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate'
INFO:cfclient.utils.zmq_led_driver:ZMQ led disabled in config file
INFO:cfclient.utils.zmq_param:ZMQ param disabled in config file
INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)]
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
Cannot find a Crazyradio Dongle
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for open_toolboxes
INFO:cfclient.ui.tab_toolbox:No config found for open_toolboxes
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
`

saaraswata, 2024-08-01:
I have the same issue still waiting for an answer

saaraswata, 2024-08-03:
Okay so I found out the whole solution, I'll be uploading a guide on YouTube later, I'll share link. Be sure to check it

lovefool, 2024-08-05:
Thanks,I am looking forward to see your solution.

closed 2024-08-13 by zhanzhaocheng

leeebo, 2024-08-14:
@lovefool The bug has already been fixed, please update the latest code and install the new PC client refer https://github.com/qljz1993/crazyflie-clients-python.git

qurrpurr, 2024-09-25:
hello I am having the same problem, installed the new cfclient but nothing works. can someone tell me the proper instruction?

leeebo, 2024-09-25:
@qurrpurr Please refer this guide https://github.com/leeebo/crazyflie-clients-python/blob/master/README.md

qurrpurr, 2024-09-25:
I have done per the instructions connected to drone via wifi but it still could not find the drone when i pressed scan.
[image]

leeebo, 2024-09-25:
@qurrpurr Please try to re-install  cflib with `pip install git+https://github.com/leeebo/crazyflie-lib-python.git`

qurrpurr, 2024-09-27:
it works now! thank you!