espressif/esp-drone #67: I have attached Ms5611 barometer to esp32s, how to enable altitude hold?

saaraswata, 2024-08-01:


leeebo, 2024-08-13:
refer: #34

saaraswata, 2024-08-13:
I have ported ms5611 successfully and it shows that ms5611 I2C connection [OK]...
However when I go to cfclient sensor plotter tab and check the barometer plotter, it seems that it's not using ms5611 for getting altitude data. When I rise the drone, the plotter goes up and then goes down to the same level. It's using mpu6050 to detect acceleration. I have a joystick and I tried to do altitude hold from cfclient too, but when I do it the drone either very slowly moved up or very slowly moved down for he target altitude.  How to tell the drone to make use of the barometer instead of accelerometer. Because on startup , barometer shows correct pressure, temp and altitude in serial monitor