espressif/esp-drone #84: cfcilent screen stop updating after 3-5 seconds

lovefool, 2024-08-18:
**Describe the bug**
I finally got connect esp-drone and cfclient.@leeebo thank you for your advice. 
cfclient screen stop updating after 3~5 seconds. "Disconnect & Connect on cfcilent screen" make it work again, but soon stops.
When cfclient connects, Green LED (bottom-right) start blinking, and when cfclient screen stops updating Green LED become off.There are no data on ctrf sniffer screen. The connection broken after a few seconds?

**Version (please complete the following information):**
 - IDF Version: [e.g. v4.4.8, commit id]
 - SoC Chip: [e.g. ESP32-S2]
 - Hardware: [e.g. ESP32_S2_Drone_V1_2]

**To Reproduce**
everytime

**Expected behavior**
**Logs**

leeebo, 2024-08-18:
@lovefool would you please add your logs from both esp32-s2 and cfclient ?

lovefool, 2024-08-19:
cfclient log
=========================
UserWarning: Using SDL2 binaries from pysdl2-dll 2.24.0
INFO:cfclient.gui:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2']
INFO:cfclient.utils.input.inputreaders:Could not initialize [linuxjsdev]: Only supported on Linux
INFO:cfclient.utils.input.inputreaders:Successfully initialized [pysdl2]
INFO:cfclient.utils.input.inputinterfaces:Found interfaces: ['leapmotion', 'wiimote', 'zmqpull']
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named 'leapsdk')
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named 'cwiid'
INFO:cfclient.utils.config:Dist config read from z:\esp-drone-leebo\crazyflie-clients-python-master\src\cfclient/configs/config.json
INFO:cfclient.utils.config:Config file read from [C:\Users\jayte.HP-PAVILION\AppData\Local\Bitcraze\cfclient/config.json]
INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file
INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate'
INFO:cfclient.utils.zmq_led_driver:ZMQ led disabled in config file
INFO:cfclient.utils.zmq_param:ZMQ param disabled in config file
INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)]
WARNING:cfclient.ui.dialogs.bootloader:Failed to make web request to get firmware-release
WARNING:cfclient.ui.dialogs.bootloader:Failed to parse firmware-releases in web request
Cannot find a Crazyradio Dongle
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cflib.crazyflie:Callback->Connection initialized[udp://192.168.43.42:2390]
INFO:cflib.crazyflie:We are connected[udp://192.168.43.42:2390], request connection setup
INFO:cflib.crazyflie.platformservice:Request _request_protocol_version()
INFO:cflib.crazyflie:Callback->Connected to [udp://192.168.43.42:2390]
INFO:cflib.crazyflie.platformservice:_crt_service_callback
INFO:cflib.crazyflie.platformservice:Request protocol version
INFO:cflib.crazyflie.platformservice:_platform_callback
INFO:cflib.crazyflie.platformservice:Protocol version (platform): 4
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
WARNING:cflib.crazyflie.toccache:Could not save cache, no writable directory
INFO:cflib.crazyflie:Log TOC finished updating
INFO:cflib.crazyflie.mem:0 memories found
INFO:cflib.crazyflie:Memories finished updating
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
WARNING:cflib.crazyflie.toccache:Could not save cache, no writable directory
INFO:cflib.crazyflie:Param TOC finished updating
INFO:cflib.crazyflie:Callback->Connection setup finished [udp://192.168.43.42:2390]
INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Attitude.json]
INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Attitude_rate.json]
INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Position.json]
INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Velocity.json]
INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning_components/Pitch.json]
[... 54 lines cut ...]
INFO:cflib.crazyflie:Closing link
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cflib.crazyflie:Callback->Disconnected from [udp://192.168.43.42:2390]
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices
INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices

esp-drone log
===============
I (74935) wifi:new:<6,0>, old:<6,0>, ap:<6,0>, sta:<255,255>, prof:6
[... cut ...]
I (78372) wifi:<ba-add>idx:2 (ifx:1, 02:1a:74:b9:59:83), tid:0, ssn:39, winSize:64
[0;32mI (80648) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.43.43[0m
[0;32mI (145843) WIFI_UDP: udp packet cksum unmatched[0m
[0;32mI (166178) WIFI_UDP: udp packet cksum unmatched[0m
[0;32mI (169370) WIFI_UDP: udp packet cksum unmatched[0m
[0;32mI (182289) WIFI_UDP: udp packet cksum unmatched[0m
[0;32mI (184305) WIFI_UDP: udp packet cksum unmatched[0m
[0;32mI (195553) WIFI_UDP: udp packet cksum unmatched[0m

lovefool, 2024-08-19:
@leebo I have added log from both system. 

Esp-drone gives "WIFI_UDP: udp packet cksum unmatched" error.

leeebo, 2024-10-08:
Hi @lovefool, have you tried the latest [esp-drone code](https://github.com/espressif/esp-drone/commits/master/) and [cfclient](https://github.com/leeebo/crazyflie-clients-python/commits/master/)?

It works for me without such error

lovefool, 2024-10-08:
Hi @leeebo, I have installed new cfclient, cflib and compiled esp-drone around Aug 19th. Then I can connect for a few seconds and got  "WIFI_UDP: udp packet cksum unmatched" error. Since then I have not updated programs. I may re-install/compile all codes.

cranberrysoft, 2024-10-20:
@lovefool I got the same error when the Drone firmware message queue was full and just crashed.

I guess the same happens to you. In general, this line in the firmware would 

```
            // We should never drop packet
            printf("CRTP RX queue full\n");
            printf("Port: %d\n", p.port);
            ASSERT(0);
```

You can try to check the logs for your drone to see if this happens to you. In my case, one of the modules related to memory was not working correctly and was not initiated so when UDP message was coming the module did not send back the response. Then the Python controller would re-send the UDP packet until the queue was full and crashed the firmware with assert(0)

So I commented on `systemWaitStart` as bellow

```
static void memTask(void* param) {
	crtpInitTaskQueue(CRTP_PORT_MEM);
 printf("Init memTask \n");

  // systemWaitStart();

 printf("Init memTask finished \n");
```

The module waited forever even though the system had been fully initiated so it was some glitch in firmware could not sort out.

jobitjoseph, 2024-11-18:
> @lovefool I got the same error when the Drone firmware message queue was full and just crashed.
> 
> I guess the same happens to you. In general, this line in the firmware
> 
> ```
>             // We should never drop packet
>             printf("CRTP RX queue full\n");
>             printf("Port: %d\n", p.port);
>             ASSERT(0);
> ```
> 
> You can try to check the logs for your drone to see if this happens to you. In my case, one of the modules related to memory was not working correctly and was not initiated so when UDP message was coming the module did not send back the response. Then the Python controller would re-send the UDP packet until the queue was full and crashed the firmware with assert(0)
> 
> So I commented on `systemWaitStart` as bellow
> 
> ```
> static void memTask(void* param) {
> 	crtpInitTaskQueue(CRTP_PORT_MEM);
>  printf("Init memTask \n");
> 
>   // systemWaitStart();
> 
>  printf("Init memTask finished \n");
> ```
> 
> The module waited forever even though the system had been fully initiated so it was some glitch in firmware could not sort out but this hack works...
> 
> Still, Python app would disconnect after a while but did not have time to continue investigating the problem...

could you please tell me which file did you edit?

jobitjoseph, 2024-11-22:
> **Describe the bug** I finally got connect esp-drone and cfclient.@leeebo thank you for your advice. cfclient screen stop updating after 3~5 seconds. "Disconnect & Connect on cfcilent screen" make it work again, but soon stops. When cfclient connects, Green LED (bottom-right) start blinking, and when cfclient screen stops updating Green LED become off.There are no data on ctrf sniffer screen. The connection broken after a few seconds?
> 
> **Version (please complete the following information):**
> 
> * IDF Version: [v4.4.8, commit id]
> * SoC Chip: [ESP32-S2]
> * Hardware: [ESP32_S2_Drone_V1_2]
> 
> **To Reproduce** everytime
> 
> **Expected behavior** **Logs**
> 
> # cfclient log
> UserWarning: Using SDL2 binaries from pysdl2-dll 2.24.0 INFO:cfclient.gui:Disabling STL printouts INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2'] INFO:cfclient.utils.input.inputreaders:Could not initialize [linuxjsdev]: Only supported on Linux INFO:cfclient.utils.input.inputreaders:Successfully initialized [pysdl2] INFO:cfclient.utils.input.inputinterfaces:Found interfaces: ['leapmotion', 'wiimote', 'zmqpull'] INFO:cfclient.utils.input.inputinterfaces:Could not initialize [leapmotion]: Leap Motion library probably not installed (No module named 'leapsdk') INFO:cfclient.utils.input.inputinterfaces:Could not initialize [wiimote]: Missing cwiid (wiimote) driver No module named 'cwiid' INFO:cfclient.utils.config:Dist config read from z:\esp-drone-leebo\crazyflie-clients-python-master\src\cfclient/configs/config.json INFO:cfclient.utils.config:Config file read from [C:\Users\jayte.HP-PAVILION\AppData\Local\Bitcraze\cfclient/config.json] INFO:cfclient.utils.input.inputinterfaces:Could not initialize [zmqpull]: ZMQ input disabled in config file INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate' INFO:cfclient.utils.zmq_led_driver:ZMQ led disabled in config file INFO:cfclient.utils.zmq_param:ZMQ param disabled in config file INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)] WARNING:cfclient.ui.dialogs.bootloader:Failed to make web request to get firmware-release WARNING:cfclient.ui.dialogs.bootloader:Failed to parse firmware-releases in web request Cannot find a Crazyradio Dongle INFO:cflib.drivers.cfusb:Looking for devices.... INFO:cflib.crtp.prrtdriver:Initialized PRRT driver. INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cflib.crazyflie:Callback->Connection initialized[udp://192.168.43.42:2390] INFO:cflib.crazyflie:We are connected[udp://192.168.43.42:2390], request connection setup INFO:cflib.crazyflie.platformservice:Request _request_protocol_version() INFO:cflib.crazyflie:Callback->Connected to [udp://192.168.43.42:2390] INFO:cflib.crazyflie.platformservice:_crt_service_callback INFO:cflib.crazyflie.platformservice:Request protocol version INFO:cflib.crazyflie.platformservice:_platform_callback INFO:cflib.crazyflie.platformservice:Protocol version (platform): 4 INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices WARNING:cflib.crazyflie.toccache:Could not save cache, no writable directory INFO:cflib.crazyflie:Log TOC finished updating INFO:cflib.crazyflie.mem:0 memories found INFO:cflib.crazyflie:Memories finished updating INFO:cfclient.utils.input.inputreaders.pysdl2:Looking for devices INFO:cfclient.utils.input.inputreaders.pysdl2:Found 0 devices WARNING:cflib.crazyflie.toccache:Could not save cache, no writable directory INFO:cflib.crazyflie:Param TOC finished updating INFO:cflib.crazyflie:Callback->Connection setup finished [udp://192.168.43.42:2390] INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Attitude.json] INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Attitude_rate.json] INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Position.json] INFO:cfclient.utils.logconfigreader:Parsing [PID_tuning/Velocity.json] INFO:cfclient.utils.logconfigreader:Parsin [... cut ...] 22 I (6207) SYSLOAD: 1.85 4592 wifi 23 I (6211) SYSLOAD: 2.04 1900 SENSORS 6 I (6216) SYSLOAD: 0.00 2096 sys_evt 20 I (6220) SYSLOAD: Free heap: 57740 bytes I (6478) SENSORS: isBiasValueFound! I (6479) STAB: Ready to fly.

There is something wrong with the UDP driver.
to fix it either
1. Uninstall the current cflib, and install the cflib from https://github.com/jobitjoseph/crazyflie-lib-python
2. Uninstall cfclient and install cfclient form https://github.com/jobitjoseph/crazyflie-clients-python

lovefool, 2024-11-24:
@jobitjoseph, thank you for your fix. I have uninstall/install cflib/clclient as below. It works better but GREEN LED stop blinking after 3~4 mins (not seconds). There is NO error on PC side cfcient. But I couldn't get esp-dron log yet. ( I am working on it.)

As for communication with iOS app, it works w/o problem. 

cfclient             0.dev1713     Z:\ESP-Drone\crazyflie-clients-python\src
cflib                0.1.27.1.dev0

love

jotheesh1729, 2025-03-25:
@leeebo @cranberrysoft 
when I am connecting to the UDP, I get this error: Failed to connect on udp://192.168.43.42:2390: Couldn't load link driver: [Errno 56] Socket is already connected,

cranberrysoft, 2025-03-27:
@jotheesh1729 
If you use MacOS  do comment out code as below in the crazyfile lib
```
        self.socket.bind(('', 2399))
        # self.socket.connect(self.addr)
```

It seems like bind, at least for MacOS implementation, also "connects". Connect seems to try to bind to the port again which ends up with this error `Socket is already connected`