espressif/esp-drone #85: Esp drone power problem

sasha240100, 2024-09-14:
Hi, I'm trying to build a drone (esp32c3-based) inspired by an esp-drone project and I've got a problem with power supply.
Whenever the motors start spinning more than ~20% of their max power, **the esp32c3 keeps restarting** (shutdown). 

I am using a **3.7v Fullymax, 600mah battery rated 70C** (which should be enough for a small quadrocopter).

My assumption that drone keeps restarting is based on blue LED flickering, and that it reconnects to wifi everytime when it happens. _**I assume that the problem is that esp32c3 isn't getting enough voltage**_ once motors start consuming more power, but I have no idea how to fix it while still staying on a 3.7v 1S battery.

I was hoping to find the answer over the internet and in the schematics of the drones from documentation:
https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-drone/en/latest/_static/ESPlane_FC_V1/Schematic_ESPlane_FC_V1.pdf
https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-drone/en/latest/_static/ESPlane_V2_S2/SCH_ESPlane_V2_S2.pdf

But unfortunately haven't found how to fix this problem yet. Maybe you could suggest why that could happen and how it could be fixed? I'm happy to provide any details!

Thanks!

[image]

mvschaefer, 2024-09-17:
@sasha240100 - i cant make out the capacitors, motors, etc, in the schematic screen shot, but have you tried:
- apply a Vbatt gain schedule/table to the max current/torque/speed to the motor(s).

The idea is to apply a derate to the motor commands as voltage drops below current capabilities (as the batteries age, etc).  Initially populate the table with ~0.9 across the Vbatt range of interest.

Also, do you have any other batteries to test?

sasha240100, 2024-10-08:
@livigni actually yes, at least I think I've got some improvement from the last time. 

The problem I noticed was that I've used a regular diode on the +5V power line, **the voltage drop of that diode I measured was 0.7V**. In my case that was a [1N4007 SOD-323](https://www.aliexpress.com/item/33054871177.html) which I think was too much for a 1S LiPo battery (3.7v) that should handle both the motors & the control unit.

After I looked carefully 2nd time at the [official schematics](https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-drone/en/latest/_static/ESPlane_V2_S2/SCH_ESPlane_V2_S2.pdf) I noticed that instead they use a BAT60B Schottky diode, which in general has a lot lower voltage drop value (I measured as 0.2V), that is a much less compared to previous 0.7V I had. 

I didn't test it enough to confirm if the issue is now gone on my side, but quick tests showed the difference in a better direction. I will test it more on weekends to find out if the issue is now completely gone away.

> P.s. I am a new learner in microelectronics world and would love to share my findings and learn from other people building a similar project. If you would like to exchange experience, drop me an email at alexbuzin88@gmail.com & let's get connected

sasha240100, 2024-10-08:
@mvschaefer Thanks for looking at my problem! The issue is definitely related to the voltage drop like you have mentioned.
I wonder are there any general recommendations on how to optimise the PCB to reduce voltage drop to minimum? 

It feels like 3.7V battery (even rated 70C) is not enough to handle such load, maybe only when fully charged up to 4.2V. But at the same time what if battery voltage drops to let's say 3.5-3.6V ? The problem will appear again.

I was thinking about voltage step-up converters to boost voltage to somewhat stable 5V, but that feels like a dumb idea. Not sure what's the best practice is for similar drones. ?

livigni, 2024-10-20:
that diode should not impact as intervenes only when usb is plugged, not when you power from battery.

In my case I thought it could be the battery which I don't know what discharge rating is (it's a 2000mAh but discharge capacity is not specified), but since it's not working for you too makes me think shouldn't be the case. there must be something wrong in the schematic/components but I didn't find it yet and lost a bit of traction over this project.

jobitjoseph, 2025-01-21:
I would suggest you change your 3.3V LDO. We had similar issues. The drone works fine with LDO from Microchip(MIC5219). But when we used MIC5219 from another brand we had similar issues. I guess there is excessive noise on the LDO's output causing these reboots. using the original MIC5219 from Microchip fixed the issue.