espressif/esp-drone #89: [Altitude Hold][MS5611 Is detected and outputs pressure value on log but drone still uses velocity based altitude/height hold]

saaraswata, 2024-09-27:
**Describe the bug**
Ms5611 is detected and gives appropriate pressure value. I have implemented it from HukeIOT repository. But it seems that the drone still uses velocity based altitude/height hold due to which drone either very slowly climbs to the ceiling or goes down to the floor while using height hold/altitude hold.

**Version (please complete the following information):**
 - IDF Version: 4.4.8
 - SoC Chip: ESP32 Wroom 32D
 - Hardware: ESP 32 Devkit C

**To Reproduce**
Steps to reproduce the behavior:
1. Go to cfclient ,attach joystick and use alt/height hold, also watch the sensor plotter for the barometer 
2. Modified on 
4. See error

**Expected behavior**
Barometer gives height based on pressure

**Logs**
Normal Logs nothing special.

**Error**
The plotter shows changes when the move th drone up or down and then goes back to normal which means it's using imu for velocity based height hold and not the barometer.

leeebo, 2024-10-08:
@saaraswata Hi, Barometer data currently does not participate in the final calculations.

saaraswata, 2024-10-13:
Which file should I be looking at to configure the drone to use baro data as well?