espressif/esp-drone #91: How to solder the motors on the print

Bodata, 2024-10-02:
Hi, 

I bought an assembled drone.
When putting it together i fail to find any documentation about the soldering of the motors.

I have 2 motors with a red and a  blue wire and i have 2 motors with a black and a white wire.

I understand 2 motors have to turn clockwise and 2 motors have to counterclockwise.
I also understand these have to be diagonal so motor 1 and 3 turn the same direction and motor 2 and 4 also.

I guess red should be soldered to positive and black to negative.
But it is not clear to me where to put which motor.

Is there any documentation how to solder the wires ?

Regards,
Peter

leeebo, 2024-10-08:
@Bodata Motor manufacturers usually provide different colored motor wires for convenience when assembling drones. But the internal structure of your motor is not different, and it supports both clockwise and counterclockwise.. 

You can first solder your motors arbitrarily and observe whether it rotates according to the diagram below. If not, please switch the + - wires of the motor.

[image]

https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-drone/en/latest/gettingstarted.html#propeller-direction

Bodata, 2024-11-04:
Thanks, i got it to work.
I have used a 1,5  battery to find the correct wires.

closed 2024-11-04 by Bodata