espressif/esp-drone #93: Flow deck for EPS32 drone

Bodata, 2024-11-04:
Hi,

I have been looking around to find documenation about extention boards.
I do not seem to find any.
In the documenation the modules Postion-hold, pressure and compass are mentioned.

Are these available somewhere, our should you make them yourself ?

All i found was Flow deck v2 of bitcraze, but the picture gives 10 pins on each size the eps32 drone has 8 pins.
[here](https://store.bitcraze.io/collections/decks/products/flow-deck-v2)

Regards,
Peter

carrliitos, 2025-03-12:
@Bodata -- Were you able to figure this out?

Bodata, 2025-03-12:
Hi,

Sorry to say, i did not get any response, so it is still not clear to me how this should work.

I have got the drone assembled but also i could not get the communication running.

Regards,
Peter

carrliitos, 2025-03-12:
For the extension boards, you do need to purchase them separately. The only issue is that the sizes of the pins of the ESP drone itself is different than what you'll see in Amazon. I bought a PMW3901 Optical Flow Sensor + VL53l0X Distance Sensor only to find out that the IO pins on the ESP drone are backwards, so the sensor module is basically useless.

- Here is the Amazon link of the exact one I bought: https://www.amazon.com/dp/B0817Q63BR?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

Bodata, 2025-03-12:
Carrliitos,

I am not sure i get your remark correctly....
So the board you bought from amazon is useless: you wrote:

 _so the sensor module is basically useless_

 But at the begining you wrote:

_For the extension boards, you do need to purchase them separately_

Is there a fitting extension board to buy or should we make it ourselfs ?

regards,
Peter

carrliitos, 2025-03-13:
@Bodata -- sorry the confusion. The board I bought from amazon (and any other boards that are currently available) was useless because the IO pin configurations on the ESP drone does not match any of the available extension boards commercially available.
I've not found an extension board that fits, so far, so it does look like we make it ourselves.

Bodata, 2025-03-15:
Carrliitos,
Thanks for your messsage.
What i do not get: there is an extension board documented but it seems never been build.
The board you bought perhaps works, but the pin layout is wrong. 
Succes building you own board.
Regards,
Peter

closed 2025-03-18 by zhanzhaocheng