espressif/esp-drone #95: Missing `idfSelectedId` Key in `idf_env_json` Configuration

carrliitos, 2024-11-12:
**Describe the bug**
When running the `idf_tools.py` script to install ESP-IDF tools via `./install.sh` command, the installation fails with a `KeyError` on the `idfSelectedId` key in `idf_env_json`. This error occurs because the `idf_env_json` configuration dictionary does not contain the expected `idfSelectedId` key. The script attempts to retrieve this key in the `get_user_defined_targets` function but encounters an issue when it is absent, resulting in a crash.

**Version (please complete the following information):**
 - IDF Version: v4.4
 - SoC Chip: ESP32-S2
 - Hardware: ESP32-S2-Drone V1.2

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout `v4.4` and update submodules by running: 
```
git checkout v4.4
git submodule update --init --recursive
```
2. Run the `install.sh` file.
3. Observe the traceback error indicating a `KeyError` on `idfSelectedId`.

**Expected behavior**
The script should either handle the missing `idfSelectedId` key gracefully or ensure that the `idf_env_json` configuration includes it by default.

**Actual Behavior**
The script crashes due to a `KeyError` for the `idfSelectedId` key, halting the tool installation process.

**Logs**
Below is the actual error message:

```
Detecting the Python interpreter
Checking "python" ...
Python 3.10.12
"python" has been detected
Installing ESP-IDF tools
Traceback (most recent call last):
  File "/home/proj/esp/esp-idf/tools/idf_tools.py", line 1828, in <module>
    main(sys.argv[1:])
  File "/home/proj/esp/esp-idf/tools/idf_tools.py", line 1824, in main
    action_func(args)
  File "/home/proj/esp/esp-idf/tools/idf_tools.py", line 1420, in action_install
    targets = clean_targets(args.targets)
  File "/home/proj/esp/esp-idf/tools/idf_tools.py", line 1083, in clean_targets
    export_targets_to_idf_env_json(targets_from_tools_json)
  File "/home/proj/esp/esp-idf/tools/idf_tools.py", line 1051, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/home/proj/esp/esp-idf/tools/idf_tools.py", line 1097, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'

```

leeebo, 2024-11-13:
This is an ESP-IDF 4.4 environment installation issue, like https://github.com/espressif/esp-idf/issues/9837

Please run `git checkout release/v4.4` to choose the latest v4.4 branch instead of the history `tag v4.4`, for the difference please refer https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/versions.html

carrliitos, 2024-11-13:
Thank you -- that worked. Closing this ticket now.

closed 2024-11-13 by carrliitos