espressif/esp-drone #102: Kalman estimator does not work after flowdeck v2 integration

saikrishnarao2, 2025-01-16:
I tried to fly with the app. Immediately after the connect, motors starts spinning randomly, unable to take off.

leeebo, 2025-01-16:
@saikrishnarao2 In position hold mode, the throttle stick controls the z-axis position. At 50%, it holds altitude; above 50%, it ascends; below 50%, it descends. Upon connection, the motor will start adjusting, especially on uneven ground. Ensure your surroundings are safe, then push the throttle directly to above 50% to let the aircraft take off. Once it reaches the target altitude, set the throttle stick back to 50%.

saikrishnarao2, 2025-01-17:
> [@saikrishnarao2](https://github.com/saikrishnarao2) In position hold mode, the throttle stick controls the z-axis position. At 50%, it holds altitude; above 50%, it ascends; below 50%, it descends. [...]

Its not working. Changed the throttle from 0 to 50, 0 to 100. No result. As spon as I click connect button in the app on the top right, motors starts spinning and does not react to any stick movement.