espressif/esp-drone #104: Motors Not Responding to Throttle Inputs After Successful App Connection

maximagination1, 2025-01-23:
Hi there. I am encountering an issue where the output pins on my ESP to my drone's motors I defined in menuconfig remain at 0V at all times even when I apply throttle stick inputs after successfully connecting to the drone via the ESP-drone app on my Android phone. Although the app shows that it has connected to the drone, there is no measurable voltage output to the motor pins (GPIO 1-4 on the XIAO ESP32-S3) when throttle joystick inputs are applied.

These are my Environment Details:
- **Board:** Seeed Studio [XIAO ESP32-S3 Sense](https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/)  
- **Firmware:** Espressif's original esp-drone firmware built using ESP-IDF
- **ESP-IDF Version:** v5.0.7  
- **App Used:** [ESP-drone app](https://github.com/EspressifApps/ESP-Drone-Android) on Android
- **Connection Type:** USB for uploading firmware, Wi-Fi connection to the app  

What steps I took:
1. Installed ESP-IDF v5.0.7 and set up the environment as per the [official documentation](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s2/get-started/index.html#get-started-get-prerequisites).  
2. Uploaded the firmware to the XIAO ESP32-S3 using the ESP-IDF.  
3. Connected to the drone via the ESP-drone app using Wi-Fi (app reported successful connection but does the same even when not connected to its dedicated AP network).  
4. Attempted to control the motors using throttle stick inputs from the app with no luck.  
5. Installed cfclient via the IDF's Powershell and ran it, but couldn't connect to the ESP32S3 board due to error: [WinError 10051] (Couldn't load link driver).

I see that there isn't any any measurable voltage output on the programmed motor pins (GPIO 1-4) when throttle inputs are applied.
The app indicates a successful connection, but I am unsure if the connection is fully functional - if any commands are even being sent... 

Below are the relevant errors/warnings observed in the monitor log:

1. **Deprecated ADC Driver:**  
   ```
   W (355) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
   ```
   *(I understand this is a warning, but including it in case it is impacting functionality.)*

2. **CPU Initialization Warning:**  
   ```
   0x40375630: call_start_cpu1 at C:/Espressif/frameworks/esp-idf-v5.0.7/components/esp_system/port/cpu_start.c:147
   ```
   (This error may indicate an issue with core initialization, but I am uncertain if it is related to the motor output problem.)

The troubleshooting steps I took:
1. Verified that the app successfully connects to the drone.  
2. Measured output voltages on the GPIO pins assigned to motors (GPIO 1-4) but observed no changes.  
3. Checked motor control parameters in the configuration file (`config.json`), and they appear correctly set:
   - `max_thrust`: 80  
   - `min_thrust`: 25  
   - `slew_limit`: 45  
4. Confirmed that the firmware compiles and uploads successfully without fatal errors.  

I would greatly appreciate any **guidance** or **suggestions** on the following:
1. Could the CPU initialization warning or deprecated ADC driver impact the motor outputs?  
2. How can I further confirm whether the app is fully connecting to the drone?  
3. Are there additional steps I can take to debug this issue (e.g., specific configuration or firmware changes)?  

Thanks in advance,

Max

maximagination1, 2025-03-17:
> no impact I think
> please make sure you follow the steps
> , and the led blink as expected

Thanks for your reply, but I resolved this issue long ago by fixing the
instability of my MPU6050.

Cheers.

closed 2025-03-18 by leeebo

tamilarasan0427, 2025-04-02:
I think I have the same problem. During boot, all motors spin in sequence for 1 second. The app connects successfully, but the motors do not respond to throttle, yaw, or other inputs. How did you fix this IMU instability issue?

Vaibhav2012-blk, 2025-09-28:
@maximagination1 hey max i used your code but i am also facing same issue how did you fix that issue please help  if you can