espressif/esp-drone #105: [Bug][Descriptions] MPU6050 I2C latch up situation (reboot, sometimes cyclically)

dvarkin-ru, 2025-02-09:
On my ESP32-S2-Drone with a mid-charged (3.7V) or low-charged battery an error may occur when receiving data from the MPU6050. Then ESP32-S2 reboots. Wifi disconnects. And sometimes the SCL line can be grounded by the MPU6050 itself, so ESP32-S2 cant generate START condition, so now ESP32-S2 restarts cyclically until the battery is disconnected.

**Version:**
 - IDF Version:  v4.4.8-dirty
 - SoC Chip: ESP32-S2
 - Hardware: ESP32-S2-Drone V1.2

**To Reproduce**
Steps to reproduce the behavior:
1. Make an error on I2C bus.

**Expected behavior**
Software restarts I2C bus. When SDA latched-up to ground, ESP32 sends some pulses on SCL.

**Logs when ESP32-S2 cyclically reboots**
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400235bd
0x400235bd: esp_restart_noos_dig at C:/esp-idf-v4.4.8/components/esp_system/esp_system.c:53 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x18d0
load:0x4004c000,len:0xd2c
load:0x40050000,len:0x30c0
entry 0x4004c23c
I (24) boot: ESP-IDF v4.4.8-dirty 2nd stage bootloader
I (24) boot: compile time 15:25:34
I (24) boot: chip revision: v0.0
I (27) qio_mode: Enabling default flash chip QIO
I (32) boot.esp32s2: SPI Speed      : 80MHz
I (37) boot.esp32s2: SPI Mode       : QIO
I (42) boot.esp32s2: SPI Flash Size : 2MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00100000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=1f5d4h (128468) map
I (120) esp_image: segment 1: paddr=0002f5fc vaddr=3ffc8670 size=00a1ch (  2588) load
I (121) esp_image: segment 2: paddr=00030020 vaddr=40080020 size=8d1d0h (578000) map
I (225) esp_image: segment 3: paddr=000bd1f8 vaddr=3ffc908c size=03474h ( 13428) load
I (228) esp_image: segment 4: paddr=000c0674 vaddr=40022000 size=16668h ( 91752) load
I (262) boot: Loaded app from partition at offset 0x10000
I (262) boot: Disabling RNG early entropy source...
I (274) cpu_start: Unicore app
I (274) cache: Instruction cache        : size 8KB, 4Ways, cache line size 32Byte
I (276) cpu_start: Pro cpu up.
I (291) cpu_start: Pro cpu start user code
I (291) cpu_start: cpu freq: 240000000
I (291) cpu_start: Application information:
I (296) cpu_start: Project name:     ESPDrone
I (301) cpu_start: App version:      48d64e7-dirty
I (306) cpu_start: Compile time:     Feb  9 2025 17:10:39
I (312) cpu_start: ELF file SHA256:  07dc8a779e85b7c3...
I (318) cpu_start: ESP-IDF:          v4.4.8-dirty
I (324) cpu_start: Min chip rev:     v0.0
I (328) cpu_start: Max chip rev:     v1.99
I (333) cpu_start: Chip rev:         v0.0
I (338) heap_init: Initializing. RAM available for dynamic allocation:
I (345) heap_init: At 3FFDD960 len 0001E6A0 (121 KiB): DRAM
I (351) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (357) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (364) spi_flash: detected chip: generic
I (368) spi_flash: flash io: qio
W (372) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (386) cpu_start: Starting scheduler on PRO CPU.
I (401) PLATFORM: set active config
I (401) gpio: GPIO[7]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (409) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (418) gpio: GPIO[9]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (429) wifi:wifi driver task: 3ffe770c, prio:23, stack:6656, core=0
I (434) system_api: Base MAC address is not set
I (439) system_api: read default base MAC address from EFUSE
I (447) wifi:wifi firmware version: ff661c3
I (449) wifi:wifi certification version: v7.0
I (453) wifi:config NVS flash: enabled
I (456) wifi:config nano formating: disabled
I (460) wifi:Init data frame dynamic rx buffer num: 32
I (465) wifi:Init static rx mgmt buffer num: 5
I (469) wifi:Init management short buffer num: 32
I (474) wifi:Init dynamic tx buffer num: 32
I (478) wifi:Init static rx buffer size: 1600
I (482) wifi:Init static rx buffer num: 10
I (486) wifi:Init dynamic rx buffer num: 32
I (490) wifi_init: rx ba win: 6
I (493) wifi_init: tcpip mbox: 32
I (497) wifi_init: udp mbox: 6
I (501) wifi_init: tcp mbox: 6
I (505) wifi_init: tcp tx win: 5760
I (509) wifi_init: tcp rx win: 5760
I (513) wifi_init: tcp mss: 1440
I (517) wifi_init: WiFi IRAM OP enabled
I (522) wifi_init: WiFi RX IRAM OP enabled
I (528) phy_init: phy_version 2500,8368a9b,Mar 21 2024,19:51:06
I (566) wifi:mode : softAP (60:55:f9:da:25:ed)
I (570) wifi:Total power save buffer number: 16
I (570) wifi:Init max length of beacon: 752/752
I (570) wifi:Init max length of beacon: 752/752
I (574) espnow: esp-now Version: 2.5.2
I (577) ESPNOW: espnow [version: 1.0] init
I (580) espnow: mac: 60:55:f9:da:25:ec, version: 2
I (586) espnow: Enable main task
I (590) espnow: main task entry
I (592) WIFI_UDP: wifi_init_softap complete.SSID:ESP-DRONE_6055F9DA25ED password:12345678
I (602) WIFI_UDP: Socket created
I (606) WIFI_UDP: Socket bound, port 2390
I (611) WIFI_UDP: UDP server create socket succeed
I (1116) SYS: ----------------------------

I (1116) SYS: ESPlane 2.0  is up and running!

I (1116) I2CDRV:  i2c 1 driver install return = 0
I (1120) EEPROM: spi_flash_init ... !
I (1124) EEPROM: spi_flash_read ok !
Characterized using Two Point Value
I (2440) I2CDRV:  i2c 0 driver install return = 0
E (2442) SENSORS: MPU6050 I2C connection [FAIL].

assert failed: sensorsDeviceInit sensors_mpu6050_hm5883L_ms5611.c:419 (0)

Backtrace: 0x40023a02:0x3ffcd8d0 0x4002bb69:0x3ffcd8f0 0x4003353e:0x3ffcd910 0x40094f0f:0x3ffcda30 0x40093605:0x3ffcda80 0x40093160:0x3ffcdaa0 0x40089ef8:0x3ffcdad0
0x40023a02: panic_abort at C:/esp-idf-v4.4.8/components/esp_system/panic.c:408

0x4002bb69: esp_system_abort at C:/esp-idf-v4.4.8/components/esp_system/esp_system.c:137

0x4003353e: __assert_func at C:/esp-idf-v4.4.8/components/newlib/assert.c:85

0x40094f0f: sensorsDeviceInit at C:/work/esp-drone/components/core/crazyflie/hal/src/sensors_mpu6050_hm5883L_ms5611.c:419
 (inlined by) sensorsMpu6050Hmc5883lMs5611Init at C:/work/esp-drone/components/core/crazyflie/hal/src/sensors_mpu6050_hm5883L_ms5611.c:680

0x40093605: sensorsInit at C:/work/esp-drone/components/core/crazyflie/hal/src/sensors.c:186

0x40093160: stabilizerInit at C:/work/esp-drone/components/core/crazyflie/modules/src/stabilizer.c:194

0x40089ef8: systemTask at C:/work/esp-drone/components/core/crazyflie/modules/src/system.c:197

ELF file SHA256: 07dc8a779e85b7c3

Rebooting...
```

leeebo, 2025-03-17:
@dvarkin-ru Thanks for your report. 

` And sometimes the SCL line can be grounded by the MPU6050 itself, so ESP32-S2 cant generate START condition, so now ESP32-S2 restarts cyclically until the battery is disconnected.`

Can you explain why MPU6050 is pulling down the SCL line? According to the specifications of the MPU6050, it should also work normally even at a voltage of 2.4v

dvarkin-ru, 2025-03-28:
I'm sorry, I mean SDA line, as I described in Expected behavior. About MPU6050, I think, it might be electromagnetic interference and/or weak pull up. Or it might be a problem with LDO. Maybe I'll look at it with an oscilloscope later. 10K pull up looks the most suspicious.

jobitjoseph, 2025-07-03:
faced similar issues with some MPU6050 clones. Change the IMU.

> On my ESP32-S2-Drone with a mid-charged (3.7V) or low-charged battery an error may occur when receiving data from the MPU6050. Then ESP32-S2 reboots. Wifi disconnects. And sometimes the SCL line can [...]

juliordzcer, 2025-07-07:
Hello everyone,

I was experiencing the exact same issue described here with my **ESP32-S2-Drone V1.2**:
- Cyclic reboots due to `MPU6050 I2C connection [FAIL]` and `assert failed: sensorsDeviceInit`.

**My Solution:**
The issue was resolved by flashing a **specific commit** of the `esp-drone` firmware. After cloning the repository and checking out to this commit, the MPU6050 now connects successfully, and the drone boots up to `STAB: Ready to fly.`.

The working commit hash is:
`d999e2010636f234229ecf9db145f6f749fcdf3c`

You can obtain this version by:
```bash
git clone [https://github.com/espressif/esp-drone.git](https://github.com/espressif/esp-drone.git)
cd esp-drone
git checkout d999e2010636f234229ecf9db145f6f749fcdf3c
```
After building and flashing this version, the MPU6050 error disappeared. (Note: I still see warnings for VL53L1X and PMW3901, but these sensors are not present on my specific hardware, so they are not relevant to the MPU6050 issue).

**Important Note**: I haven't yet tested with more recent firmware versions of esp-drone to see if the MPU6050 issue reappears. I will update this comment if I identify the root cause or a more recent version that also works.

I hope this helps others facing similar problems!

juliordzcer, 2025-07-10:
Hello everyone,

As an update to my previous message:

While I'm still unsure of the root cause of the initial error, I've discovered something interesting. After flashing the older firmware version that resolved the MPU6050 issue, I decided to update to the latest version of the `esp-drone firmware`.

To my surprise, after updating to the latest version, the MPU6050 error has not reappeared! The drone continues to boot correctly and displays "STAB: Ready to fly..".

So, it seems a potential solution (though not an explanation for the cause) is to first flash the `d999e2010636f234229ecf9db145f6f749fcdf3c` version and then update to the most recent esp-drone firmware version.

I hope this additional information is helpful for anyone still experiencing this problem. I'll continue to update if I find more details!