espressif/esp-drone #108: build error

linxzhang, 2025-04-08:
when I using VSCODE to build esp-drone, I encounter below error:
crtp_commander_generic.c
C:/examples/Enter/components/core/crazyflie/modules/src/crtp_commander_generic.c: In function 'fullStateDecoder':
C:/examples/Enter/components/core/crazyflie/modules/src/crtp_commander_generic.c:337:3: error: 'quatdecompress' accessing 16 bytes in a region of size 4 [-Werror=stringop-overflow=]
  337 |   quatdecompress(values->quat, (float *)&setpoint->attitudeQuaternion.q0);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/examples/Enter/components/core/crazyflie/modules/src/crtp_commander_generic.c:337:3: note: referencing argument 2 of type 'float[4]'
In file included from C:/examples/Enter/components/core/crazyflie/modules/src/crtp_commander_generic.c:36:
C:/examples/Enter/components/core/crazyflie/modules/interface/quatcompress.h:65:20: note: in a call to function 'quatdecompress'  
   65 | static inline void quatdecompress(uint32_t comp, float q[4])

leeebo, 2025-04-08:
Please build under ESP-IDF 4.4 or 5.0. Other IDF versions are not guaranteed

[image]

jvssoft, 2025-05-06:
change:  
quatdecompress(values->quat, (float *)&setpoint->attitudeQuaternion);

zghnx, 2025-06-09:
> change: quatdecompress(values->quat, (float *)&setpoint->attitudeQuaternion);

 bug?