espressif/esp-drone #116: [Bug][Descriptions]

Its-BB, 2025-07-19:
**Describe the bug**
A diff build of the code works on my esp32 drone but when i try to build using this code (both are same) it does not work, I read some and compared both the codes and i saw, 'Ready to fly' does not come and also ' I (6543) SENSORS: isBiasValueFound!' does not come. But note: **THE DRONE FULLY WORKS IN THE PREMADE-.BIN FILE**

**Version (please complete the following information):**
 - IDF Version: 4.4.5
 - SoC Chip: ESP32
 - Hardware: ESPlane

**Logs**
Add logs to help explain your problem.
I (27) boot: ESP-IDF v4.4.5-dirty 2nd stage bootloader
I (27) boot: compile time 00:09:03
I (27) boot: chip revision: v3.1
I (31) qio_mode: Enabling default flash chip QIO
I (36) boot.esp32: SPI Speed      : 80MHz
I (41) boot.esp32: SPI Mode       : QIO
I (45) boot.esp32: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1fb6ch (129900) map
I (137) esp_image: segment 1: paddr=0002fb94 vaddr=3ffb0000 size=00484h (  1156) load
I (138) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=904c0h (591040) map
I (306) esp_image: segment 3: paddr=000c04e8 vaddr=3ffb0484 size=03c44h ( 15428) load
I (310) esp_image: segment 4: paddr=000c4134 vaddr=40080000 size=1784ch ( 96332) load
I (354) boot: Loaded app from partition at offset 0x10000
I (354) boot: Disabling RNG early entropy source...
I (366) cpu_start: Pro cpu up.
I (366) cpu_start: Starting app cpu, entry point is 0x400813dc
0x400813dc: call_start_cpu1 at C:/Users/ItsBB/Desktop/esp-idf-v4.4.5/components/esp_system/port/cpu_start.c:147

I (353) cpu_start: App cpu up.
I (382) cpu_start: Pro cpu start user code
I (382) cpu_start: cpu freq: 160000000
I (382) cpu_start: Application information:
I (387) cpu_start: Project name:     ESPDrone
I (392) cpu_start: App version:      1
I (396) cpu_start: Compile time:     Jul 20 2025 00:08:51
I (403) cpu_start: ELF file SHA256:  4dc16297cf7a3210...
I (408) cpu_start: ESP-IDF:          v4.4.5-dirty
I (414) cpu_start: Min chip rev:     v0.0
I (419) cpu_start: Max chip rev:     v3.99
I (423) cpu_start: Chip rev:         v3.1
I (428) heap_init: Initializing. RAM available for dynamic allocation:
I (435) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (441) heap_init: At 3FFCA960 len 000156A0 (85 KiB): DRAM
I (448) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (454) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (460) heap_init: At 4009784C len 000087B4 (33 KiB): IRAM
I (467) spi_flash: detected chip: generic
I (471) spi_flash: flash io: qio
W (475) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (489) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (546) PLATFORM: set active config
I (547) gpio: GPIO[18]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (549) gpio: GPIO[23]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (558) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (570) wifi:wifi driver task: 3ffd385c, prio:23, stack:6656, core=0
I (574) system_api: Base MAC address is not set
I (579) system_api: read default base MAC address from EFUSE
I (592) wifi:wifi firmware version: 0f80fa0
I (592) wifi:wifi certification version: v7.0
I (593) wifi:config NVS flash: enabled
I (596) wifi:config nano formating: disabled
I (600) wifi:Init data frame dynamic rx buffer num: 32
I (605) wifi:Init management frame dynamic rx buffer num: 32
I (611) wifi:Init management short buffer num: 32
I (615) wifi:Init dynamic tx buffer num: 32
I (619) wifi:Init static rx buffer size: 1600
I (623) wifi:Init static rx buffer num: 10
I (627) wifi:Init dynamic rx buffer num: 32
I (631) wifi_init: rx ba win: 6
I (635) wifi_init: tcpip mbox: 32
I (639) wifi_init: udp mbox: 6
I (642) wifi_init: tcp mbox: 6
I (646) wifi_init: tcp tx win: 5744
I (650) wifi_init: tcp rx win: 5744
I (654) wifi_init: tcp mss: 1440
I (658) wifi_init: WiFi IRAM OP enabled
I (663) wifi_init: WiFi RX IRAM OP enabled
I (673) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (774) wifi:mode : softAP (28:56:2f:4a:0e:e5)
I (776) wifi:Total power save buffer number: 16
I (776) wifi:Init max length of beacon: 752/752
I (777) wifi:Init max length of beacon: 752/752
I (781) espnow: esp-now Version: 2.5.2
I (785) ESPNOW: espnow [version: 1.0] init
I (789) espnow: mac: 28:56:2f:4a:0e:e4, version: 2
I (794) espnow: Enable main task
I (798) espnow: main task entry
I (799) WIFI_UDP: wifi_init_softap complete.SSID:ESP-DRONE_28562F4A0EE5 password:12345678
I (811) WIFI_UDP: Socket created
I (815) WIFI_UDP: Socket bound, port 2390
I (819) WIFI_UDP: UDP server create socket succeed
I (1325) SYS: ----------------------------

I (1325) SYS: ESPlane 2.0  is up and running!

I (1325) I2CDRV:  i2c 1 driver install return = 0
I (1329) EEPROM: spi_flash_init ... !
I (1333) EEPROM: spi_flash_read ok !
Characterized using eFuse Vref
I (2549) I2CDRV:  i2c 0 driver install return = 0
I (2550) SENSORS: MPU6050 I2C connection [OK].

W (2906) ZR2: Z-down sensor [FAIL]

W (2906) SENSORS: VL53L1X I2C connection [FAIL].

I (2906) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
W (2961) SENSORS: PMW3901 SPI connection [FAIL].

I (2961) SENSORS: sensors init done
I (2962) SENSORS: pitch_calib = 0.000000,roll_calib = 0.000000
I (2966) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:1
I (2976) ESTIMATOR: Using Complementary (1) estimator

I (2982) POSITION_CONTROLLER: thrustBase = 36000,thrustMin  = 20000
I (2989) SYS: wifilinkTest = 1
I (3492) SYS: pmTest = 1
I (3492) SYS: workerTest = 1
I (3492) SYS: systemTest = 1
I (3492) SYS: configblockTest = 1
I (3495) COMM: wifilinkTest = 1
I (3499) COMM: crtpTest = 1
I (3502) COMM: crtpserviceTest = 1
I (3507) COMM: platformserviceTest = 1
I (3511) COMM: consoleTest = 1
I (3515) COMM: paramTest = 1
I (3519) SYS: commTest = 1
I (3522) SYS: commanderTest = 1
I (5344) SYS: stabilizerTest = 1
I (5344) SYS: estimatorKalmanTaskTest = 1
I (5344) SYS: soundTest = 1
No one-wire memory handler registered
I (5348) SYS: memTest = 1
I (5352) STAB: Wait for sensor calibration...

I (5357) SYS: systemStart ! selftestPassed = 1
I (6329) SYSLOAD: Task dump
I (6329) SYSLOAD: Load  Stack left      Name    PRI
I (6329) SYSLOAD: 0.24  2428    Tmr Svc         1
I (6331) SYSLOAD: 94.53         1560    IDLE    0
I (6335) SYSLOAD: 94.93         1556    IDLE    0
I (6340) SYSLOAD: 0.03  2584    CRTP-RX         2
I (6345) SYSLOAD: 0.23  3436    PWRMGNT         1
I (6349) SYSLOAD: 0.05  2664    tiT     18
I (6353) SYSLOAD: 0.46  3288    STABILIZER      7
I (6358) SYSLOAD: 0.01  1956    sys_evt         20
I (6363) SYSLOAD: 0.06  2212    espnow_main     1
I (6368) SYSLOAD: 0.00  3660    UDP_TX  2
I (6372) SYSLOAD: 0.00  3264    UDP_RX  2
I (6377) SYSLOAD: 0.00  3572    WIFILINK        2
I (6382) SYSLOAD: 0.00  2644    CRTP-TX         2
I (6386) SYSLOAD: 0.00  2584    LOG     2
I (6391) SYSLOAD: 0.00  1580    PARAM   2
I (6395) SYSLOAD: 0.00  1532    CMDHL   3
I (6399) SYSLOAD: 1.08  1072    ipc1    4
I (6404) SYSLOAD: 0.00  2388    KALMAN  4
I (6408) SYSLOAD: 0.00  1552    MEM     2
I (6413) SYSLOAD: 0.00  1608    LEDSEQCMD       1
I (6417) SYSLOAD: 4.01  3664    SYSTEM  1
I (6422) SYSLOAD: 0.11  4376    SENSORS         6
I (6426) SYSLOAD: 0.07  3436    esp_timer       22
I (6431) SYSLOAD: 3.10  4456    wifi    23
I (6436) SYSLOAD: 0.47  1100    ipc0    24
I (6440) SYSLOAD: Free heap: 123160 bytes

amitx64, 2025-07-20:
@Its-BB Does your CFClient work properly? I mean, does it update all the data without freezing? Mine hangs after about 3 seconds.

amitx64, 2025-07-20:
@Its-BB, I cloned the `cfclient` from Leeebo's GitHub repository. After cloning, I installed it using the following command:

```bash
pip install -e .
```

The installation completed successfully, and the CFClient opens. However, after connecting to the ESP-based drone, it freezes and doesn't update any of the data.